hypriot/flash #9: Install package pv if not installed

MathiasRenner, 2015-04-18:
The script needs pv. It does not work without.
 
Suggestion: Check if this package is available beforehand - and install if not. The installation would require sudo...

Any better ideas?

DieterReuter, 2015-04-18:
Just remove the strict dependency from using `pv`. Maybe this needs some testing and a bug fix on Linux.

StefanScherer, 2015-04-18:
@MathiasRenner I do not have a Linux test env yet, but the code looks very similar to the Mac version. The script checks if pv is installed (in path) and has both if and else implemented with and without the `pv` command.

See this code lines https://github.com/hypriot/flash/blob/master/Linux/flash#L194-L204

There is only a notice that there is no progress bar.

I wouldn't do an automatic installation of pv inside the flash script.

vielmetti, 2015-05-23:
I ran into this issue on my Mac and was able to get past it by running 
   `brew install pv`
and then proceeding.

StefanScherer, 2015-05-29:
@vielmetti thank you. That was exactly the way I've installed `pv` as well. 
I have added some installation notes in the [README.md](https://github.com/hypriot/flash/blob/master/README.md#install-dependencies).

closed 2015-05-29 by StefanScherer