hypriot/flash #15: occi and wifi not working

wallnerryan, 2015-07-23:
using flash did not add credentials correclty, manually editing the credentials files and running ```occi``` did not work and fails with ```occidentalis not installed``` and when manually trying to run the occidentalis install script that fails as well.

using fresh Elizabeth img.

StefanScherer, 2015-07-26:
How did you call the `flash` script? Can you give me an example (of course without your real data).

Try quoting the values for the SSID and the password with single quotes, eg. `--password 'My$ecret` to avoid bash interpretation of variables etc.

Do you use the Mac or Linux version of the flash script?

StefanScherer, 2015-09-09:
I'll close this one. Please let me know if you still have issues.

closed 2015-09-09 by StefanScherer