hypriot/flash #25: Request: Static ip address option

kaspernissen, 2016-01-25:
First of all, thank you for an awesome tool!

Is there anyway to setup a static ip like the option hostname? Would save some time, when building a Raspberry Pi cluster.

Thank you in advance.

StefanScherer, 2016-01-25:
@kaspernissen Thanks for your feedback. We're already thinking about some kind of "device init" that does more than the occidentalis.
This will happen with our new image built with [hypriot/image-builder-rpi](https://github.com/hypriot/image-builder-rpi).

Once the image is ready we will update `flash` to put a similar file to the occidentalis.txt into the boot partition and then we will have much more features.

StefanScherer, 2016-02-13:
@kaspernissen We have opened the `device-init` repo. Please file an issue at https://github.com/hypriot/device-init with your request ;-)

closed 2016-02-13 by StefanScherer

saphoooo, 2016-06-25:
At least, once wlan0 is configured, the ability to add persistent network configuration through a reboot.

flashydave, 2016-10-29:
It appears that on reboot device-init will currently overwrite any subsequent settings you may have in /etc/network/interfaces/wlan0 to set a static IP - so at very least some documentation as a workaround would assist people until this feature is available.

tsouhaieb, 2017-02-15:
i already found a workaround, check [device-init#6](https://github.com/hypriot/device-init/issues/6)