hypriot/flash #29: flash doesn't work with mounted .dmg file

StefanScherer, 2016-02-10:
Quick note, where should I file bugs on the flash utility. If you have a mounted .dmg file, it doesn't work correctly (for instance if you have one mounted from an install).

Reported by @richtong

StefanScherer, 2016-02-10:
Reproducable, eg. with a VirtualBox .dmg file and no SD card inserted:

```
+ df
Filesystem    512-blocks      Used Available Capacity  iused    ifree %iused  Mounted on
/dev/disk1     974700800 679981336 294207464    70% 85061665 36775933   70%   /
devfs                669       669         0   100%     1159        0  100%   /dev
map -hosts             0         0         0   100%        0        0  100%   /net
map auto_home          0         0         0   100%        0        0  100%   /home
/dev/disk4s1      194488    190960      3528    99%    23868      441   98%   /Volumes/VirtualBox
+ true
+ echo ''

+ read -p 'Is /dev/disk4s1 correct? ' yn
Is /dev/disk4s1 correct? 
```

richtong, 2016-02-10:
With a dmg and a flash it picks up both disks in the grep for disk names. I think the setting of variable disk gets two names c

Thanks,
Rich

Nothing in the world can take the place of persistence... The slogan "press on" has solved and will always solve the problems of the human race. Locomotive Engineers Monthly. November 1914.

http://tongfamily.com | @richtong | http://fb.me/richtong1 | +1-206-963-9029

closed 2016-02-13 by StefanScherer

richtong, 2016-02-13:
> Closed #29 via #39.

Thanks much. Very clean bash which is a rarity :-)

Thanks,
Rich

Nothing in the world can take the place of persistence... The slogan "press on" has solved and will always solve the problems of the human race. Locomotive Engineers Monthly. November 1914.

http://tongfamily.com | @richtong | http://fb.me/richtong1 | +1-206-963-9029

PR #39 (2016-02-13): Recognize SD card without FAT partition
Fixes #38 recognize SD card without FAT partition
fixes #29 work with mounted .dmg files

Doesn't "hang" after flashing an SD card image without FAT partition, just abort after 5 seconds and unmount everything without occidentalis.txt/config.txt features.