hypriot/flash #33: [Linux] Flashing fails when image name different from archive name

MathiasRenner, 2016-02-11:
This (see headline) holds e.g. when flashing Hypriot Cluster Lab. Then this is what happens:

```
$ ./flash hypriot_20160121-235123_clusterlab.img.zip
/usr/bin/unzip
Uncompressing /home/[...]/hypriot_20160121-235123_clusterlab.img.zip ...
Archive:  /home/[...]/hypriot_20160121-235123_clusterlab.img.zip
  inflating: /tmp/hypriot-rpi-20160121-235123.img  
Use /tmp/  hypriot-rpi-20160121-235123.img
Filesystem                  1K-blocks      Used Available Use% Mounted on
udev                          8046676         4   8046672   1% /dev
tmpfs                         1612604      1816   1610788   1% /run
[...]

Is /dev/mmcblk0 correct? y
Synchronizing Filesystems
[sudo] password for ###: 
Unmounting /dev/mmcblk0 ...
Flashing /tmp/  hypriot-rpi-20160121-235123.img to /dev/mmcblk0 ...
stat: cannot stat ‘hypriot-rpi-20160121-235123.img’: No such file or directory
cat: /tmp/: Is a directory
cat: hypriot-rpi-20160121-235123.img: No such file or directory
   0B 0:00:00 [   0B/s] [>                                     ]  0%            
0+0 records in
0+0 records out
0 bytes (0 B) copied, 6,177e-05 s, 0,0 kB/s
Flushing Buffers
Mounting Disk
Mounting /dev/mmcblk0p1 to customize
Unmounting /dev/mmcblk0 ...
Finished.
```

If archive name is identical with image name inside the archive, everything works fine.

StefanScherer, 2016-02-11:
@MathiasRenner So this https://github.com/hypriot/flash/blob/master/Linux/flash#L127-L129 doesn't work?

StefanScherer, 2016-02-11:
Oh, I see: `Flashing /tmp/  hypriot-rpi-20160121-235123.img` there are blanks in the path.

StefanScherer, 2016-02-11:
We should cherry-pick https://github.com/jeff-99/flash/commit/7f0a6b1c82acccb45ffabf7fad09b660b026584b

PR #34 (2016-02-13): fixed unzipped filename resolution to exclude spaces
Fixes #33 

@MathiasRenner please test this PR on Linux.
closed 2016-02-13 by StefanScherer