hypriot/flash #44: Can't run flash

pascalandy, 2016-03-12:
For some reasons, I can't run the flash application. Any idea why ?

[image]

Happy Saturday!

StefanScherer, 2016-03-12:
After downloading the script you have to set the executable flag
```
chmod +x flash
```

Have we missed mentioning that somewhere?

pascalandy, 2016-03-12:
I did many and followed the https://github.com/hypriot/flash#installation but still nothing.
The flash file 

```
flash               100%[===================>]   7.01K  --.-KB/s    in 0s      

2016-03-12 17:06:48 (180 MB/s) - 'flash' saved [7175/7175]

hypriot|⇒ chmod +x flash
hypriot|⇒ sudo chmod +x flash
Password:
hypriot|⇒ sudo mv flash /usr/local/bin/flash
hypriot|⇒ sudo flash --help
sudo: flash: command not found
hypriot|⇒ 

```
la -la returns
```

drwxr-xr-x  12 pascalandy  admin   408 12 Mar 17:07 .
drwxrwxr-x@ 64 root        admin  2176 12 Mar 16:37 ..
drwxr-xr-x  13 pascalandy  admin   442 12 Mar 17:10 .git
-rw-r--r--   1 pascalandy  admin    23 12 Mar 16:36 .gitignore
-rw-r--r--   1 pascalandy  admin   387 12 Mar 16:36 AUTHORS
drwxr-xr-x   3 pascalandy  admin   102 12 Mar 16:47 Darwin
-rw-r--r--   1 pascalandy  admin  1075 12 Mar 16:36 LICENSE
drwxr-xr-x   3 pascalandy  admin   102 12 Mar 16:36 Linux
-rw-r--r--   1 pascalandy  admin  5837 12 Mar 16:36 README.md
-rw-r--r--   1 pascalandy  admin   714 12 Mar 16:36 Vagrantfile
-rwxr-xr-x   1 pascalandy  admin  7175 12 Mar 17:06 flash
-rw-r--r--   1 pascalandy  admin  3095 12 Mar 16:36 sample.vmx
flash|master⚡ ⇒ 

```

StefanScherer, 2016-03-12:
Seems like your PATH environment variable does not contain `/usr/local/bin`.
You can check this with `echo $PATH`.
Try `/usr/local/bin/flash --help`.

Otherwise after downloading it to the current directory try use `./flash`.

pascalandy, 2016-03-12:
```
echo $PATH
> /usr/local/bin:/bin:/sbin:/usr/bin:/usr/sbin:/usr/libexec
```

and also 

```
flash|master⚡ ⇒ sudo /usr/local/bin/flash --help
sudo: /usr/local/bin/flash: command not found
```

I'll try current directory :)

StefanScherer, 2016-03-12:
As the absolute path does not work I guess the current directory also does not work.

The flash script uses `/bin/bash` as intepreter.

I have seen that you use zsh. Is bash installed? Try `which bash` and `/bin/bash` and then  `exit` from the bash again.

pascalandy, 2016-03-12:
Your rock Stefan! Here is the conclusion.

```
cd some-folder
wget https://raw.githubusercontent.com/hypriot/flash/master/Darwin/flash
chmod +x ./flash
sudo ./flash --help
```

The system return: https://gist.github.com/pascalandy/92e457179af0a91ac55b
Let's put my first micro SD card.

`sudo ./flash --hostname swrpi3-master01 hypriot-rpi-20160306-192317.img`

StefanScherer, 2016-03-12:
Congrats! 
BTW you don't need `sudo`. The script does the sudo internally where needed.

pascalandy, 2016-03-12:
Allright!

closed 2016-03-12 by pascalandy