hypriot/flash #47: flash doesn't work with brew coreutils installed

jwhitcraft, 2016-03-17:
I have coreutils installed via bre which means dd and stat are differnet that the stock osx ones.

The follow output is what i get.

```
↪  flash --hostname pi-docker hypriot-rpi-20160306-192317.img.zip
Uncompressing hypriot-rpi-20160306-192317.img.zip ...
Archive:  hypriot-rpi-20160306-192317.img.zip
  inflating: /tmp/hypriot-rpi-20160306-192317.img
Use /tmp/hypriot-rpi-20160306-192317.img
No SD card found. Please insert SD card, I'll wait for it...
Filesystem      Size  Used Avail Use% Mounted on
/dev/disk1      465G  409G   57G  88% /
/dev/disk6s1    7.4G  4.2M  7.4G   1% /Volumes/PI

Is /dev/disk6 correct?
Please answer yes or no.

Is /dev/disk6 correct? yes
Unmounting disk6 ...
Unmount of all volumes on disk6 was successful
Unmount of all volumes on disk6 was successful
Flashing /tmp/hypriot-rpi-20160306-192317.img to disk6 ...
Password:
stat: cannot read file system information for '%B': No such file or directory
pv: -s: integer argument expected
dd: invalid number: ‘1m’
Unmounting and ejecting disk6 ...
Unmount of all volumes on disk6 was successful
Unmount failed for /dev/disk6s2
Disk /dev/disk6 ejected
🍺  Finished.
```

I've fixed it by changing the paths to point directly at the programs installed in `/usr/bin` and it works fine.

StefanScherer, 2016-03-17:
@jwhitcraft Please download the script again. I've just merged #46 which should fix that.

jwhitcraft, 2016-03-17:
@StefanScherer, thanks, I'll file another ticket for when `gnu-sed` is installed

jwhitcraft, 2016-03-17:
Looks good, just need #48 fixed and it would be awesome!

closed 2016-03-17 by jwhitcraft