hypriot/flash #53: Default timezone

fxstein, 2016-03-31:
Right now the default timezone that a flashed image comes up with is `Europe/Berlin` from `/etc/timezone' and it seems to be set by '/root/firstboot.sh'

Might be a good idea to take the default timezone from the Mac/Linux system that the cards are flashed from, as that will often match the systems/cluster one wants to setup.

Could set the timezone after the flashing before unmounting and ejecting the SD card.

StefanScherer, 2016-03-31:
Thanks. Marked the idea in device-init repo as well.

StefanScherer, 2017-11-29:
We have switched to cloud-init with HypriotOS 1.7.x so please have a look at our latest blog post about using cloud-init or the example /boot/user-data in the latest SD card image. There is a comment how to use/set the time zone.

closed 2017-11-29 by StefanScherer