hypriot/flash #56: Change script name to hypriot-flash

fxstein, 2016-04-02:
Considering broader deployment options like home-brew as in #55 , it might be a good idea to rename the script from `flash` to `hypriot-flash` to avoid current and future naming conflicts.

RnbWd, 2016-04-03:
but there are no naming conflicts haha (flash is dead)

fxstein, 2016-04-03:
Call me paranoid. `flash` is a very common term, and its very specific to `hypriot`. as simple `hy` prefix would do. If not I can live with that as well, just a suggestion.

RnbWd, 2016-04-04:
np I'm just joking around :stuck_out_tongue_winking_eye: --- There's a cask and a science tap named flash - but nothing in the official repo. I like `hy` - anything shorter than `hypriot`. I always misspell the name and I use flash for all raspi images, I think the library's unique and deserves an awesome namespace. But it doesn't really matter - I'd alias it anyway.

fxstein, 2016-04-04:
Anyway its really up to @StefanScherer and his team to consider. As I said just an opinion.

StefanScherer, 2016-06-08:
Let's keep the name short. Closing.

closed 2016-06-08 by StefanScherer

witsch, 2020-04-03:
Fwiw, I've created https://github.com/Homebrew/homebrew-core/pull/52503 to hopefully soon be able to run `brew install flash`.  Let's hope the name will be acceptable for the homebrew maintainers… 🙂

StefanScherer, 2020-04-16:
It just landed, thanks @witsch !!! 🎉🎉  I couldn't believe this was possible 😅 

[image]