hypriot/flash #60: Command to display the script's version

jmMeessen, 2016-05-07:
In my opinion, a command should be available (ex: `--version` or `-v`) to display the version of the script.

StefanScherer, 2016-05-09:
@jmMeessen Great idea.
I'll remove the `-v` option from the `--verbose` and display the version instead.

I thought how to maintain that version string in both scripts, and still support pull requests from the community. 
* For these bash scripts we will go with a GitHub release that inserts the version into both scripts. 
* In the source code there will a version number like "dirty". 
* We have to update the installation steps to download it from the GitHub release page instead directly from source tree.

StefanScherer, 2017-12-17:
As we now have a small CI pipeline on Circle + Travis we could think of automating this and make releases with version numbers ;-)

StefanScherer, 2018-06-02:
Finally, you can run `flash --version`.

Download latest release with

```
curl -LO https://github.com/hypriot/flash/releases/download/2.1.0/flash
chmod +x flash
sudo mv flash /usr/local/bin/flash
```