hypriot/flash #74: adding hidden ssid supported

mrgnou, 2016-07-16:
Actually, the generated wifi configuration doesn't provide hidden ssid support.
It would be nice to manage hidden wifi networks by adding an optional param scan-ssid to device-init.yaml such as:

```
hostname: black-pearl
wifi:
  interfaces:
    wlan0:
      ssid: "MyNetwork"
      scan-ssid: 1
      password: "secret_password"
```

This way, generated wlan0 file would look like 

```
allow-hotplug wlan0

auto wlan0
iface wlan0 inet dhcp
  wpa-ssid MyNetwork
  wpa-scan-ssid 1
  wpa-psk generated_psk
```

StefanScherer, 2016-07-18:
@mrgnou Thanks, good idea. I've opened a issue in device-init which does the real work.

StefanScherer, 2017-11-29:
We have switched to cloud-init with HypriotOS 1.7.x so please have a look at our latest blog post about using cloud-init or the example `/boot/user-data` in the latest SD card image. As you can specify the whole wpa config you should be able to do this.

StefanScherer, 2017-12-17:
Should be manageable with a customized `user-data` file. See https://github.com/hypriot/image-builder-rpi/blob/master/builder/files/boot/user-data how the default `/boot/user-data` file of HypriotOS 1.7.x looks like and adjust it to your needs. Then use the latest flash script with option `flash --userdata your-customized-hidden-ssid-user-data-file`.

closed 2017-12-17 by StefanScherer