hypriot/flash #77: Failed to flash over existing version using defaults

mikeycmccarthy, 2016-08-22:
Hi,

Just tried to use flash 1.0.0 over a version of 0.8 that wasn't installed using flash:

```bash
~  flash --hostname XXX --ssid XXX --password XXX https://downloads.hypriot.com/hypriotos-rpi-v1.0.0.img.zip
Downloading https://downloads.hypriot.com/hypriotos-rpi-v1.0.0.img.zip ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  231M  100  231M    0     0  8231k      0  0:00:28  0:00:28 --:--:-- 10.1M
Uncompressing /tmp/image.img.zip ...
Archive:  /tmp/image.img.zip
  inflating: /tmp/hypriotos-rpi-v1.0.0.img
Use /tmp/hypriotos-rpi-v1.0.0.img
Filesystem      Size   Used  Avail Capacity  iused    ifree %iused  Mounted on
/dev/disk0s2   238Gi   75Gi  162Gi    32% 19805272 42499598   32%   /
devfs          184Ki  184Ki    0Bi   100%      638        0  100%   /dev
map -hosts       0Bi    0Bi    0Bi   100%        0        0  100%   /net
map auto_home    0Bi    0Bi    0Bi   100%        0        0  100%   /home
/dev/disk1s1    64Mi   30Mi   34Mi    47%      512        0  100%   /Volumes/HypriotOS
s /dev/disk1 correct? yes
Unmounting disk1 ...
Unmount of disk1 failed: at least one volume could not be unmounted
Unmount of disk1 failed: at least one volume could not be unmounted
Flashing /tmp/hypriotos-rpi-v1.0.0.img to disk1 ...
Password:
dd: /dev/rdisk1: Resource busy
   0 B 0:00:00 [   0 B/s] [>                                                                                                         ]  0%
Set hostname=XXX
Set wlan0/ssid=XXX
Set wlan0/password=XXX
Unmounting and ejecting disk1 ...
Unmount of disk1 failed: at least one volume could not be unmounted
Unmount of disk1 failed: at least one volume could not be unmounted
Volume failed to eject
👎  Someting went wrong.
```

At I guess is it because of how the card has been formatted?

davidjmurray, 2016-08-22:
The error message states that a partition/volume could  not be unmounted.  This can be due  to a folder  open on that volume  or you are accessing that volume. Try closing all folders  or open  windows and try again.

mikeycmccarthy, 2016-08-22:
All working now, thanks!

closed 2016-08-22 by mikeycmccarthy