hypriot/flash #80: `umount` fails if existing partition has a space in the name

r24y, 2016-08-29:
I think it probably just needs quotes in the `umount` section:

```
ryan@washington [09:53:28 PM] [~] 
-> % flash -d /dev/sdc backup.img 
NAME             SIZE TYPE  MOUNTPOINT
sda            238.5G disk  
├─sda1           200M part  /boot/efi
├─sda2         152.6G part  
├─sda3         619.9M part  
├─sda4           3.7G part  
│ └─cryptswap1   3.7G crypt [SWAP]
└─sda5          81.4G part  /
sdc              7.5G disk  
├─sdc1           1.7G part  /media/ryan/Linux Mint 18 MATE 64-bit
└─sdc2           2.3M part  

Is /dev/sdc correct? yes
Unmounting /dev/sdc ...
umount: 64-bit: mountpoint not found
```

r24y, 2016-08-29:
For now, I was able to continue by manually unmounting:

```
umount /dev/sdc1
umount /dev/sdc2
```

StefanScherer, 2016-08-29:
@r24y Can you test it again by changing `$NF` to `$1` line 209 https://github.com/hypriot/flash/blob/f36dbcc3c1914b1719cecd729f2fd89e5096ae0d/Linux/flash#L209 
If this fixes your problem a PR would be very welcome, don't forget to edit the AUTHORS file :-)

r24y, 2016-08-30:
Definitely! I'll test it today or tomorrow; my Linux box unexpectedly needed rebuilding.

georgezero, 2016-09-20:
I can confirm that changing `$NF `to `$1` on line 209, fixes the issue when the space in the volume name issue.

Thanks for the awesome script.

StefanScherer, 2016-09-21:
@georgezero Thanks for confirming. Fixed with https://github.com/hypriot/flash/commit/6828a335ef85aaee62e005affd168c933ab43536

commit 6828a335 (2016-09-21): Fix umount with space in partition name. Fixes #80
closed 2016-09-21 by StefanScherer