hypriot/flash #83: Customizing after flashing fails

MathiasRenner, 2016-09-25:
When I run
```
./flash --hostname somenode hypriotos-rpi-v1.0.1.img
```
... the flashing itself is successful, but the customizing afterwards is not. I get the error
**"Permission denied"**

Does anyone else run into this issue?

### Details
**The terminal output:**
```
Flashing hypriotos-rpi-v1.0.1.img to /dev/sdb ...
1000MiB 0:00:00 [1,71GiB/s] [================================>] 100%            
24+7681 records in
24+7681 records out
1048576000 bytes (1,0 GB, 1000 MiB) copied, 42,5457 s, 24,6 MB/s
Waiting for device...
/dev/sdb: Permission denied
```

**Environment:**
  - OS: Linux, Ubuntu 16.04
  - Same error with different SD->MicroSD adapters (lock is not enabled, of course) as well as with an USB 3.0->MicroSD adapter

StefanScherer, 2016-10-16:
Probably fixed with #88 ? Otherwise try to run the script with `set -x` to see which command does not have enough privileges.

StefanScherer, 2017-12-17:
I cannot reproduce it with latest version of Linux/flash, maybe fixed with #88 and/or #106.

closed 2017-12-17 by StefanScherer