hypriot/flash #89: permission error after  waiting for device...

dkebler, 2016-10-12:
Flash wrote out the image ok despite the error and I booted right up!....

this permission error appears awhile after the waiting for device.. echo

not sure what flash is doing at that point but obviously not critical to getting a usable card.

using on ubunutu 14.04.

[image]

MathiasRenner, 2016-10-12:
I can confirm this error, same for me! (Ubuntu 16.04.1)
It happens for me with any adapter (USB -> MicroSD or Standard SD card adapter)
Thereby, the write protection is disabled of course.

StefanScherer, 2017-12-17:
Maybe fixed with #106, I tried the latest version in a Linux VM and can't see an error.
Closing this now. Feel free to re-open if the problem still exists on your machine with the latest Linux/flash script.

closed 2017-12-17 by StefanScherer