hypriot/flash #90: May be good to check /sys/block/XXX/removable

martingkelly, 2016-10-17:
This isn't a bug, but it is a suggestion that I think would be helpful.

One safety check that could be added to this script is to enumerate only devices that show as removable so that system disks and similar are not even suggested as options for flashing. This can be done on Linux by reading the file */sys/block/XXX/removable*, where *XXX* is your device name (e.g. *sda*).

TohtoriKuka, 2017-12-21:
Or mayde it's even better to read /sys/block/ directory before and after when user plugs the removable media to get the exact name of device.

StefanScherer, 2019-03-03:
Closing due to inactivity. If someone wants to come up with a PR I'm happy to merge. I don't use flash on Linux so PR's should come from the community

closed 2019-03-03 by StefanScherer