hypriot/flash #93: Set static IP for eth0

henriklied, 2016-11-23:
I see there's a setting for connecting to a specified WLAN, but is there also a way to set a static IP adress for eth0?

StefanScherer, 2016-11-23:
No. All these features lead to extending device-init with more and more tests involved. At the moment we don't have the time to enhance this. Maybe we should look at cloud-init which is able to do more things.

StefanScherer, 2017-11-29:
We've switch to cloud-init with HypriotOS 1.7.x so you should be able to provide a `--userdata my-user-data` file. Have a look at http://cloudinit.readthedocs.io/en/0.7.9/index.html if this is possible with cloud-init / cloud-config.
The `flash` script only provides the option to copy the `user-data` file into the boot section.

closed 2017-11-29 by firecyberice