hypriot/flash #96: SD Card device (issue?)

sbeeker, 2017-03-05:
I have used this tool before on previous releases of images to flash SD cards.
Since I upgraded MAC to Sierra I am getting a bad device.  

Not sure if this is a problem or if I have screwed up something on mac.  The SD cards do not seem to be getting written properly at all anymore.

Not sure why the change in device name to rdisk2 occurs or why the DD is failing.  

pogo:flash-master sbeeker$ sudo flash --hostname pi1 https://blog.hypriot.com/hypriotos-rpi-v1.1.3.img.zip
Downloading https://blog.hypriot.com/hypriotos-rpi-v1.1.3.img.zip ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 11329    0 11329    0     0  58897      0 --:--:-- --:--:-- --:--:-- 59005
No SD card found. Please insert SD card, I'll wait for it...
Filesystem                          Size   Used  Avail Capacity iused      ifree %iused  Mounted on
/dev/disk1                         931Gi  741Gi  189Gi    80% 2811341 4292155938    0%   /
devfs                              186Ki  186Ki    0Bi   100%     642          0  100%   /dev
map -hosts                           0Bi    0Bi    0Bi   100%       0          0  100%   /net
map auto_home                        0Bi    0Bi    0Bi   100%       0          0  100%   /home
map -fstab                           0Bi    0Bi    0Bi   100%       0          0  100%   /Network/Servers
localhost:/fDQm93gxSUdjdppJeemvo8  931Gi  931Gi    0Bi   100%       0          0  100%   /Volumes/MobileBackups
/dev/disk2s1                        30Gi  1.9Mi   30Gi     1%       0          0  100%   /Volumes/PI

Is /dev/disk2 correct? y
Unmounting disk2 ...
Unmount of all volumes on disk2 was successful
Unmount of all volumes on disk2 was successful
Flashing /tmp/image.img.zip to disk2 ...
11.1KiB 0:00:00 [62.1MiB/s] [================================>] 100%            
dd: /dev/rdisk2: Invalid argument
0+1 records in
0+0 records out
0 bytes transferred in 0.000399 secs (0 bytes/sec)
Unmounting and ejecting disk2 ...
Unmount of all volumes on disk2 was successful
Unmount failed for /dev/disk2s2
Disk /dev/disk2 ejected
🍺  Finished.

StefanScherer, 2017-03-05:
@sbeeker I'm using Sierra and don't have such problems.

When I list the raw devices before plugging in a SD card it looks like this

```
$ ls -ld /dev/rdisk*
crw-r-----  1 root    operator    1,   0 Feb 27 15:58 /dev/rdisk0
crw-r-----  1 root    operator    1,   1 Feb 27 15:58 /dev/rdisk0s1
crw-r-----  1 root    operator    1,   3 Feb 27 15:58 /dev/rdisk0s2
crw-r-----  1 root    operator    1,   2 Feb 27 15:58 /dev/rdisk0s3
crw-r-----  1 root    operator    1,   4 Feb 27 15:58 /dev/rdisk1
cr--r-----  1 stefan  staff       1,   5 Mar  1 09:12 /dev/rdisk3
cr--r-----  1 stefan  staff       1,   7 Mar  1 09:12 /dev/rdisk3s1
cr--r-----  1 stefan  staff       1,   6 Mar  1 09:12 /dev/rdisk3s2
```

And after inserting a SD card there is /dev/rdisk2

```
$ ls -ld /dev/rdisk*
crw-r-----  1 root    operator    1,   0 Feb 27 15:58 /dev/rdisk0
crw-r-----  1 root    operator    1,   1 Feb 27 15:58 /dev/rdisk0s1
crw-r-----  1 root    operator    1,   3 Feb 27 15:58 /dev/rdisk0s2
crw-r-----  1 root    operator    1,   2 Feb 27 15:58 /dev/rdisk0s3
crw-r-----  1 root    operator    1,   4 Feb 27 15:58 /dev/rdisk1
crw-r-----  1 root    operator    1,   8 Mar  5 08:37 /dev/rdisk2
crw-r-----  1 root    operator    1,   9 Mar  5 08:37 /dev/rdisk2s1
crw-r-----  1 root    operator    1,  10 Mar  5 08:37 /dev/rdisk2s2
cr--r-----  1 stefan  staff       1,   5 Mar  1 09:12 /dev/rdisk3
cr--r-----  1 stefan  staff       1,   7 Mar  1 09:12 /dev/rdisk3s1
cr--r-----  1 stefan  staff       1,   6 Mar  1 09:12 /dev/rdisk3s2
```

The "dd" command on my machine is here:

```
$ which dd
/bin/dd

$ ls -l /bin/dd
-rwxr-xr-x  1 root  wheel  31952 Dec 10 10:04 /bin/dd
```

sbeeker, 2017-03-05:
I run same commands you did output below... I am using 32gb SD cards.

Last login: Sun Mar  5 06:42:10 on ttys002
pogo:~ sbeeker$ ls -ld /dev/rdisk*
crw-r-----  1 root  operator    1,   0 Mar  5 04:30 /dev/rdisk0
crw-r-----  1 root  operator    1,   1 Mar  5 04:30 /dev/rdisk0s1
crw-r-----  1 root  operator    1,   3 Mar  5 04:30 /dev/rdisk0s2
crw-r-----  1 root  operator    1,   2 Mar  5 04:30 /dev/rdisk0s3
crw-r-----  1 root  operator    1,   4 Mar  5 04:30 /dev/rdisk1
crw-r-----  1 root  operator    1,  10 Mar  5 05:42 /dev/rdisk4
crw-r-----  1 root  operator    1,  12 Mar  5 05:42 /dev/rdisk4s1
crw-r-----  1 root  operator    1,  11 Mar  5 05:42 /dev/rdisk4s2
pogo:~ sbeeker$ ls -ld /dev/rdisk*
crw-r-----  1 root  operator    1,   0 Mar  5 04:30 /dev/rdisk0
crw-r-----  1 root  operator    1,   1 Mar  5 04:30 /dev/rdisk0s1
crw-r-----  1 root  operator    1,   3 Mar  5 04:30 /dev/rdisk0s2
crw-r-----  1 root  operator    1,   2 Mar  5 04:30 /dev/rdisk0s3
crw-r-----  1 root  operator    1,   4 Mar  5 04:30 /dev/rdisk1
crw-r-----  1 root  operator    1,   5 Mar  5 07:16 /dev/rdisk2
crw-r-----  1 root  operator    1,   6 Mar  5 07:16 /dev/rdisk2s1
crw-r-----  1 root  operator    1,  10 Mar  5 05:42 /dev/rdisk4
crw-r-----  1 root  operator    1,  12 Mar  5 05:42 /dev/rdisk4s1
crw-r-----  1 root  operator    1,  11 Mar  5 05:42 /dev/rdisk4s2
pogo:~ sbeeker$ which dd
/bin/dd
pogo:~ sbeeker$ ls -l /bin/dd
-rwxr-xr-x  1 root  wheel  31952 Dec 10 04:04 /bin/dd
pogo:~ sbeeker$

sbeeker, 2017-03-05:
Prior to running command I did a reformat using the command...

sudo diskutil eraseDisk FAT32 RASPBIAN MBRFormat /dev/disk2

I have fetched the "flash" several times, chmod the permissions and moved it to /usr/local/bin per the instructions.

StefanScherer, 2017-03-05:
Maybe something with your sudo settings?
The flash script uses sudo dd inside the script. What if you remove that in the script and you run sudo flash as shown above?

sbeeker, 2017-03-05:
I removed the sudo from the script. Ran it with "sudo flash". Again the same results. It downloads the image, but will not do the dd command with same issue.

sbeeker, 2017-03-06:
Stefan,  I figured out my user error.  The host has since changed. I was able to download some images from the source host name I used but some were not working.   It appeared like it was still downloading the image but for the ones I was doing the checksums were not coming down causing no file etc.

I actually think I have the swarm working but am really not sure how to verify this.    When I do a docker -machine info I get all these machines in swarm but not running?  The commands have changed, etc. since the articles were done.    I was able to run 10 pings but not sure if they were across all the 5 pi's I have. I guess I am not sure what I have done enough to understand what I can do or want to do yet....

Right now I can ssh into six pis,  I ran the swarm master and 5 of the swarm joins. From the article I started I had to add the general swarm-user=pirate and I had to manually ssh into pis and fix the OS type in the os-config file   to have OS-  = debian instead of raspberry pi.     When I do a docker machine list I see all six machines.  However, I do not see any running in the docker ps or the docerk info.      I used the hyperiot 1.0.0 build since the 1.1.1. was not downloading.    Other than that I did everything per the instuctions.  

I was not sure about the master / slave config.  I did them remote from my mac?  In the slaves joining I do not see how I do anything to say whom the master is?

sbeeker, 2017-03-06:
User Error

closed 2017-03-06 by sbeeker