hypriot/flash #97: Unable to execute flash command on openSUSE

Doctor-Who, 2017-03-30:
Hi,

When I try to flash and install Hypriot on SD card from openSUSE Tumbleweed, terminal answer : 

`flash hypriotos-rpi-v1.4.0.img.zip 
Using cached image /tmp/hypriotos-rpi-v1.4.0.img
Unknown OS: linux
`
Any idea ?

firecyberice, 2017-03-30:
Please have a look at [L160-L178](https://github.com/hypriot/flash/blob/master/Linux/flash#L160-L178).
The output of `uname -s` seems to bee linux on SUSE instead of Linux like it is on debian.

Doctor-Who, 2017-03-31:
Thx for your reply,

Command returning me Linux : 

`trenzalore@zenbook:~> uname -s
Linux
`

StefanScherer, 2017-12-09:
@Doctor-Who 
Hm, it seems that OSTYPE is already set, but is "linux" (lower case).
I've fixed that in https://github.com/hypriot/flash/pull/110
Let me know if you still have problems.

closed 2017-12-09 by StefanScherer