hypriot/flash #100: Improve docs or fix parameter order

iago1460, 2017-07-20:
If the extra parameters are at the end, like in unix style, are ignored:
`flash hypriotos-rpi-v1.4.0.img.zip --device disk4 --config device-init.yaml --bootconf config.txt`
device, config and bootconf are being ignored without raising any warning. 

The current way is:
`flash --device disk4 --config device-init.yaml --bootconf config.txt hypriotos-rpi-v1.4.0.img.zip`
which is a bit weird.

Thanks

StefanScherer, 2017-07-20:
Yes, options after the SD card does not work, but that's "standard" in the `getopts` we use in the script. If you know another way, let me know.

So showing the usage in https://github.com/hypriot/flash#usage didn't help you?
> $ flash --help
> usage: /Users/stefan/bin/flash [options] name-of-rpi.img

iago1460, 2017-07-20:
I missed that bit, but I still think that raising some kind of error/warning if any parameter is not processed would be ideal.

StefanScherer, 2018-03-22:
Closing due to inactivity.

closed 2018-03-22 by StefanScherer