hypriot/flash #109: --ssid and --password options do not work

fudekun, 2017-11-25:
I fail to connect to the wireless LAN with the --ssid option and the --password option. hypriot is version 1.7.1, flash is the latest version. The same is true when using device-init.yaml. Please help me.

StefanScherer, 2017-11-26:
Thanks. Yes this does not work at the moment. With Hypriot 1.7.0 we have switched from device-init to cloud-init and now there is a `boot/user-data` file.
Seems like `flash` should write these two options into [user-data write_files section](https://github.com/hypriot/image-builder-rpi/blob/6277f851d469510a02b7475230453d5a515dee7e/builder/files/boot/user-data#L56-L57).

fudekun, 2017-11-29:
Thank you for your answer.
In the case of the URL method taught to you, I can not use "package-*" related description of "cloud-init". It is very inconvenient.
I need the options device - init.yml and --ssid and - password.

joevandyk, 2018-08-23:
@StefanScherer should the documentation be updated?

pisuke, 2018-10-30:
I've also experimented the same problem. 

Should the correct way to specify the wi-fi settings be to use the write_files section and add a custom wpa_supplicant.conf file put inside /etc/wpa_supplicant?

Here is a wpa_supplicant configuration that works for me with the 1.9.0 version of HypriotOS.

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=UK

network={
	ssid="SSID"
	psk="PASSWORD"
}

I agree that the user documentation should be updated.

StefanScherer, 2020-04-01:
Fixed with #165, but a prepared user-data file must be used with uncommented section.

closed 2020-04-01 by StefanScherer