hypriot/flash #118: missing "/dev/" prefix in autodetect_device (linux)

kaluzki, 2018-04-05:
```bash
$ flash hypriotos-rpi-v1.8.1-rc1.img 

Is mmcblk0 correct? y
mmcblk0: No such file or directory
Unmounting mmcblk0 ...
The SD card is write protected. Please eject, remove protection and insert again.

Is mmcblk0 correct? ^C

$ uname -a
Linux dell 4.4.0-116-generic #140-Ubuntu SMP Mon Feb 12 21:23:04 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

Woile, 2018-05-18:
Had the same issue, solved specifying `--device /dev/mmcblk0`. For someone else running into it, you can check your devices doing `lsblk`

tMatSuZ, 2018-05-18:
Hi StefanScherer
This problem occurs when the host PC is Linux. Because of this problem, if you are using flash in other projects, it will not work on Linux as it is. It works on Mac ...
I would like to see my pull request.

StefanScherer, 2018-05-19:
There's a much simpler PR #124, I haven't the time yet to check on a Linux system.