hypriot/flash #125: Does not copy user-data yml

Madic-, 2018-05-03:
Hello,

I'm using the flash utility to flash an sdcard with a userdata yml file with the following command:
flash -d /dev/mmcblk0 -f hypriotos-rpi-v1.9.0.img -u node01.yml

But the node01.yml file does not get copied to /boot/user-meta.
It's still the default user-meta. The syntax is correct.
If I manually copy the file to /boot/user-meta and boot the rpi, everything is working

Log Output from the flash command:
```
(ansible2.4) [mne-adm@tp013 hypriot]$ flash -d /dev/mmcblk0 -f hypriotos-rpi-v1.9.0.img
[sudo] password for mne-adm: 
Unmounting /dev/mmcblk0 ...
Flashing hypriotos-rpi-v1.9.0.img to /dev/mmcblk0 ...
1000MiB 0:00:00 [1.86GiB/s] [===================================================================================================================================================================>] 100%            
242+4866 records in
242+4866 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 78.2925 s, 13.4 MB/s
Waiting for device /dev/mmcblk0

/dev/mmcblk0:
 re-reading partition table
Mounting Disk
Mounting /dev/mmcblk0 to customize...
total 35294
drwxr-xr-x. 3 mne-adm mne-adm    16384 Jan  1  1970 .
drwxrwxr-x. 3 mne-adm mne-adm       60 May  3 10:15 ..
-rwxr-xr-x. 1 mne-adm mne-adm    22264 Apr 22 17:33 bcm2708-rpi-0-w.dtb
-rwxr-xr-x. 1 mne-adm mne-adm    21761 Apr 22 17:33 bcm2708-rpi-b.dtb
-rwxr-xr-x. 1 mne-adm mne-adm    22020 Apr 22 17:33 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x. 1 mne-adm mne-adm    21474 Apr 22 17:33 bcm2708-rpi-cm.dtb
-rwxr-xr-x. 1 mne-adm mne-adm    23044 Apr 22 17:33 bcm2709-rpi-2-b.dtb
-rwxr-xr-x. 1 mne-adm mne-adm    24240 Apr 22 17:33 bcm2710-rpi-3-b.dtb
-rwxr-xr-x. 1 mne-adm mne-adm    24503 Apr 22 17:33 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x. 1 mne-adm mne-adm    22952 Apr 22 17:33 bcm2710-rpi-cm3.dtb
-rwxr-xr-x. 1 mne-adm mne-adm    52064 Mar 28 14:07 bootcode.bin
-rwxr-xr-x. 1 mne-adm mne-adm      240 Apr 28 20:57 cmdline.txt
-rwxr-xr-x. 1 mne-adm mne-adm      203 Apr 28 20:57 config.txt
-rwxr-xr-x. 1 mne-adm mne-adm    18693 Apr 22 16:27 COPYING.linux
-rwxr-xr-x. 1 mne-adm mne-adm     2599 Apr 17 13:50 fixup_cd.dat
-rwxr-xr-x. 1 mne-adm mne-adm     6575 Apr 17 13:50 fixup.dat
-rwxr-xr-x. 1 mne-adm mne-adm     9726 Apr 17 13:50 fixup_db.dat
-rwxr-xr-x. 1 mne-adm mne-adm     9730 Apr 17 13:50 fixup_x.dat
-rwxr-xr-x. 1 mne-adm mne-adm 13172908 Apr 22 17:33 kernel7.img
-rwxr-xr-x. 1 mne-adm mne-adm 10245764 Apr 22 17:33 kernel.img
-rwxr-xr-x. 1 mne-adm mne-adm     1494 Mar  9 19:28 LICENCE.broadcom
-rwxr-xr-x. 1 mne-adm mne-adm       23 Apr 28 20:53 meta-data
-rwxr-xr-x. 1 mne-adm mne-adm      396 Apr 28 21:01 os-release
drwxr-xr-x. 2 mne-adm mne-adm    12288 Apr 28 20:56 overlays
-rwxr-xr-x. 1 mne-adm mne-adm   673444 Apr 17 13:50 start_cd.elf
-rwxr-xr-x. 1 mne-adm mne-adm  4968292 Apr 17 13:50 start_db.elf
-rwxr-xr-x. 1 mne-adm mne-adm  2825124 Apr 17 13:50 start.elf
-rwxr-xr-x. 1 mne-adm mne-adm  3912164 Apr 17 13:50 start_x.elf
-rwxr-xr-x. 1 mne-adm mne-adm     2050 Apr 28 20:53 user-data
Unmounting /dev/mmcblk0 ...
Finished.
```

Kind Regards,
   Madic

StefanScherer, 2018-05-12:
Wow. This is strange. You could try to run it with adding `set -x` in the second line of the script. 
Maybe we can find out what‘s going wrong with that output.

StepFischer, 2018-07-18:
HI,

same on my site.

```
$ flash --userdata ./user-data.yaml hypriotos-rpi-v1.9.0-default.img.zip 
/usr/bin/unzip
Uncompressing hypriotos-rpi-v1.9.0-default.img.zip ...
Archive:  hypriotos-rpi-v1.9.0-default.img.zip
  inflating: /tmp/hypriotos-rpi-v1.9.0.img  
Use /tmp/hypriotos-rpi-v1.9.0.img
NAME     SIZE TYPE MOUNTPOINT
sda    465,8G disk 
├─sda1  93,1G part /
sdb    465,8G disk 
├─sdb1   450M part 
├─sdb2    99M part /boot/efi
├─sdb3    16M part 
└─sdb4 465,2G part 
sdc      1,8T disk 
├─sdc1   128M part 
└─sdc2   1,8T part 
sdd      1,8T disk 
└─sdd1   1,8T part 
sde    149,1G disk 
└─sde1   149G part /run/media/stephan/Xterne Platte
├─sda2  14,9G part [SWAP]
└─sda3 357,7G part /home
sdg     59,7G disk 
├─sdg1    64M part 
└─sdg2   935M part /run/media/stephan/root
sr0     1024M rom  
Please pick your device: sdg

Is /dev/sdg correct? yes
Unmounting /dev/sdg ...
Flashing /tmp/hypriotos-rpi-v1.9.0.img to /dev/sdg ...
1000MiB 0:00:00 [2,69GiB/s] [===================================================================================================================================================================>] 100%            
666+1570 records in
666+1570 records out
1048576000 bytes (1,0 GB, 1000 MiB) copied, 51,6436 s, 20,3 MB/s
Waiting for device /dev/sdg

/dev/sdg:
 re-reading partition table
Mounting Disk
Mounting /dev/sdg to customize...
total 35294
drwxr-xr-x 3 stephan users    16384  1. Jan 1970  .
drwxr-xr-x 3 stephan users       60 18. Jul 08:08 ..
-rwxr-xr-x 1 stephan users    22264 22. Apr 17:33 bcm2708-rpi-0-w.dtb
-rwxr-xr-x 1 stephan users    21761 22. Apr 17:33 bcm2708-rpi-b.dtb
-rwxr-xr-x 1 stephan users    22020 22. Apr 17:33 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 stephan users    21474 22. Apr 17:33 bcm2708-rpi-cm.dtb
-rwxr-xr-x 1 stephan users    23044 22. Apr 17:33 bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 stephan users    24240 22. Apr 17:33 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 stephan users    24503 22. Apr 17:33 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 stephan users    22952 22. Apr 17:33 bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 stephan users    52064 28. Mär 14:07 bootcode.bin
-rwxr-xr-x 1 stephan users      240 28. Apr 20:57 cmdline.txt
-rwxr-xr-x 1 stephan users      203 28. Apr 20:57 config.txt
-rwxr-xr-x 1 stephan users    18693 22. Apr 16:27 COPYING.linux
-rwxr-xr-x 1 stephan users     2599 17. Apr 13:50 fixup_cd.dat
-rwxr-xr-x 1 stephan users     6575 17. Apr 13:50 fixup.dat
-rwxr-xr-x 1 stephan users     9726 17. Apr 13:50 fixup_db.dat
-rwxr-xr-x 1 stephan users     9730 17. Apr 13:50 fixup_x.dat
-rwxr-xr-x 1 stephan users 13172908 22. Apr 17:33 kernel7.img
-rwxr-xr-x 1 stephan users 10245764 22. Apr 17:33 kernel.img
-rwxr-xr-x 1 stephan users     1494  9. Mär 19:28 LICENCE.broadcom
-rwxr-xr-x 1 stephan users       23 28. Apr 20:53 meta-data
-rwxr-xr-x 1 stephan users      396 28. Apr 21:01 os-release
drwxr-xr-x 2 stephan users    12288 28. Apr 20:56 overlays
-rwxr-xr-x 1 stephan users   673444 17. Apr 13:50 start_cd.elf
-rwxr-xr-x 1 stephan users  4968292 17. Apr 13:50 start_db.elf
-rwxr-xr-x 1 stephan users  2825124 17. Apr 13:50 start.elf
-rwxr-xr-x 1 stephan users  3912164 17. Apr 13:50 start_x.elf
-rwxr-xr-x 1 stephan users     2050 28. Apr 20:53 user-data
Copying cloud-init ./user-data.yaml to /tmp/1000/mnt.21123/user-data ...
Unmounting /dev/sdg ...
Finished.
```

`/boot` is still empty after flash. Using `set -x` prints nothing in the terminal.

Any suggestions?

StefanScherer, 2019-03-03:
Does this still happen with the latest version of flash? I assume you do this on Linux?
I can't really help here.

g0lg0th7, 2019-08-07:
Hi, this thread help me.
In my case, running fedora (vmbox, usb), I had difficulties to mount sdcard after dd step.
Have :
- disabled selinux
- installed dos filesystem utilities from yum
- installed pv and hdparm

Work well after
Thanks