hypriot/flash #128: Fails to run inside Docker container

breandan, 2018-06-20:
Running `flash` inside a Docker container with the command ` docker run -it --privileged <docker-image> flash <some_image>` fails with the following output:

```
NAME          SIZE TYPE MOUNTPOINT
loop0       364.8M loop 
loop2       365.2M loop 
loop3        86.6M loop 
loop4       591.1M loop 
loop5       143.9M loop 
loop6       361.3M loop 
loop7       565.6M loop 
loop8        93.5M loop 
loop9        86.9M loop 
loop10      564.8M loop 
sdc          28.7G disk 
`-sdc1       28.7G part 
sdd          59.5G disk 
|-sdd1         64M part 
`-sdd2       59.4G part 
nvme0n1       477G disk 
|-nvme0n1p1   499M part 
|-nvme0n1p2   100M part 
|-nvme0n1p3    16M part 
|-nvme0n1p4  48.2G part 
|-nvme0n1p5 412.3G part /etc/hosts
`-nvme0n1p6  15.9G part 
Please pick your device: sdd

Is /dev/sdd correct? y
Unmounting /dev/sdd ...
Flashing /tmp/hypriotos-rpi-v1.9.0.img to /dev/sdd ...
1000MiB 0:00:00 [1002MiB/s] [====================================================================================>] 100%            
661+1962 records in
661+1962 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 0.98766 s, 1.1 GB/s
Waiting for device /dev/sdd

/dev/sdd:
 re-reading partition table
 BLKRRPART failed: Device or resource busy
```

Any suggestions for how to get this to work?

StefanScherer, 2018-06-20:
I have no idea if this works. We run the bats integration tests inside a Docker container, but only to "flash" a normal file inside the container. https://github.com/hypriot/flash/blob/a5ab4e5abf8623401f7453338fc46e464290ba35/Makefile#L13

StefanScherer, 2019-03-03:
Closing due to inactivity.

closed 2019-03-03 by StefanScherer