hypriot/flash #129: Add option to resize root partition

breandan, 2018-06-26:
It would be convenient to have an option to set the root partition size, so that users do not need to rebuild the HypriotOS image.

StefanScherer, 2018-06-26:
I don't quite understand this. The root filesystem will be resized on first boot. So what's the problem?
Flash couldn't help here I guess.

breandan, 2018-06-26:
The motivation here is to support putting larger files on `/var/local`, as it is currently [limited to 1GB](https://github.com/hypriot/image-builder-raw/blob/master/builder/rpi/build.sh#L12). After flashing an image, and prior to inserting the SD card into the RPI, I would like to write a file, however the default root allocation is not large enough. I would like to avoid running `image-builder-rpi`. Maybe this is not a common use case, although it seems fairly standard.

StefanScherer, 2018-06-27:
I don‘t know if that can be done with flash, changing Linux disk size from a Mac will be hard. 
If you find a way you are welcome to send a PR.

StefanScherer, 2019-03-03:
Closing due to inactivity.

closed 2019-03-03 by StefanScherer

QuantumLove, 2020-08-05:
I would love this to be possible as well.
I have got some code for Linux to accomplish it but after resizing the image and copying the files it is bricked.

@breandan did you ever get it to work?

<details><summary>CODE</summary>

echo "Image preparation script has started"
set -euf -o pipefail

baseImage=
imageFile=
deployPackage=
userDataFile=

rm $imageFile
sudo rm /tmp/$imageFile # TODO - Remove this, only applies to debuging
cp $baseImage $imageFile

partedInfo="$(sudo parted --machine "${imageFile}" print)"
sectorSize="$(echo "${partedInfo}" | grep :file: | awk -F ':' '{ print $4; }')"

startSectors=( $(fdisk --list --bytes "${imageFile}" | tail -n 2 | awk '{ print $2; }') )
bootStart=$((startSectors[0] * sectorSize))
rootStart=$((startSectors[1] * sectorSize))

dirMount="$(mktemp --directory)"
unmountImage() {
    echo "Unmounting image..."
    sleep 1
    sudo umount "${dirMount}"
}
mountRoot() {
    echo "Mounting root partition (offset=$rootStart) to ${dirMount}..."
    sleep 1
    sudo mount -o loop,offset=$rootStart "${imageFile}" "${dirMount}"
}
mountRoot
rootFree="$(sudo df "${dirMount}" | grep "${dirMount}" | awk '{ print $4; }')"
deploySize="$(du -b "${deployPackage}" | awk '{ print $1; }')"
addSize=$(( deploySize - rootFree * 1024 ))
unmountImage
rm $dirMount -R

if [ $addSize -gt 0 ] # Resize Image if needed

then
    echo "Need ${addSize} more bytes to fit ${deployPackage}"
    addSize=$(( 10 + addSize / (1000 * 1000) ))
    oldEnd="$(echo "${partedInfo}" \
        | grep '^2:' \
        | awk -F ':' '{ print $3; }' \
        | grep --color=never -oE '[0-9]+' \
    )"
    newEnd="$((oldEnd + addSize))MB"

    echo "Adding ${addSize}MB to root partition of ${imageFile}..."
    sudo dd if=/dev/zero bs=1MB count=$addSize >> "${imageFile}"

    echo "Resizing root partition to end at ${newEnd}..."
    sudo parted "${imageFile}" resizepart 2 "${newEnd}"
    sudo kpartx -a "${imageFile}"

    devLoop="$(sudo losetup -a \
        | grep `basename "${imageFile}"` \
        | grep --color=never -oP '(?<=^/dev/)loop[0-9]+' \
    )"
    devMapper="/dev/mapper/${devLoop}p2"
    sudo e2fsck -f -y -C 0 "${devMapper}"
    sudo resize2fs -p "${devMapper}"

    sleep 3     # Give time for system clean ups

    echo "Removing ${devLoop}..."
    sudo kpartx -d "${imageFile}"
    sudo dmsetup info | grep --color=never -o "${devLoop}p.*" | xargs sudo dmsetup remove || :
    sudo losetup -a | grep -q "${devLoop}:" && sudo losetup -d "/dev/${devLoop}" || :
fi

sudo guestfish --rw -a "${imageFile}"<<_EOF_
  run
  mount /dev/sda2 /
  mount /dev/sda1 /boot
  echo "Copying user-data..."
  copy-in "${userDataFile}" /boot
  echo "Upacking the home directory..."
  tgz-in "${deployPackage}" /home
_EOF_

</details>

Bazmundi, 2020-08-16:
I am all of a sudden unable to connect to my pi running hypriot nor start the docker demon and it appears that the logs are report insufficient space for some reason.  The only partition that is "full" is my root partition, so the only culprit is then the 100% full root partition.  So, if the idea is that there is no reason to allow for increasing the root partition then how does one recover from a problem like this?