hypriot/flash #131: Can't use flash on Mac (dd: invalid number: ‘1m’)

ofcourseican, 2018-07-16:
When using flash on my Mac I just get a dd error message.

Is this a bug in the script or am I doing something wrong?

```
flash --bootconf config.txt --userdata cloud-init.yml hypriotos-rpi-v1.9.0.img.zip
Using cached image /tmp/hypriotos-rpi-v1.9.0.img

Is /dev/disk2 correct? y
Unmounting /dev/disk2 ...
Unmount of all volumes on disk2 was successful
Unmount of all volumes on disk2 was successful
Flashing /tmp/hypriotos-rpi-v1.9.0.img to /dev/rdisk2 ...
dd: invalid number: ‘1m’
```

ChristianUlbrich, 2018-09-30:
@ofcourseican Which version of macOS are you running? What version of dd are you running? Do you have the stock dd? What says `which dd`?

For me flash works with 10.13.5 and the stock dd (from `/bin/dd`)

ofcourseican, 2018-10-01:
`/usr/local/opt/coreutils/libexec/gnubin/dd`

So my guess is that brew installed this.

StefanScherer, 2018-10-01:
Yes, look like the `coreutils` formula installs another `dd`. I tried it in a fresh macOS environment, but I still get the official `/bin/dd` as result from a `which dd`. Deep in the cellar I can find `/usr/local/Cellar/coreutils/8.30/libexec/gnubin/dd`. But it is not in PATH.

You probably should clean up your brew installation.

StefanScherer, 2018-10-01:
Only idea that could make it into the `flash` script is using `/bin/dd` with absolute path to stick to the official Mac binary.

StefanScherer, 2019-03-03:
Closing due to inactivity.

closed 2019-03-03 by StefanScherer