hypriot/flash #133: Writing json file using cloud-init makes user loose permissions

Jacobh2, 2018-07-30:
Hi!

I'm using Hypriot 1.9.0 and a custom could-init file to configure my RPi setup.

The yaml file looks as follows:
```
#cloud-config

hostname: qw0000p0
manage_etc_hosts: true
apt_preserve_sources_list: true

users:
  - name: client
    primary-group: users
    shell: /bin/bash
    sudo: ALL=(ALL) NOPASSWD:ALL
    groups: users,docker,adm,dialout,audio,plugdev,netdev,video
    ssh-import-id: None
    lock_passwd: true
    ssh_authorized_keys:
      - ssh-rsa A ....

locale: "en_US.UTF-8"

timezone: "Europe/Stockholm"

write_files:
  - content: |
      {
        "testkey": "testvalue",
        "mykey": "anothervalue"
      }  
    path: /home/client/setup/data.json
```

and after ssh:ing into the pi using the ssh key, I'm faced with the following:
```
> ssh client@192.168.2.2
The authenticity of host '192.168.2.2 (192.168.2.2)' can't be established.
ECDSA key fingerprint is SHA256:wpDOD9kZRTN/m4I6SEM7kvL76XEL5Ov2heeQvCFvvTw.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '192.168.2.2' (ECDSA) to the list of known hosts.
Enter passphrase for key '.ssh/id_rsa':
Linux qw0000p0 4.14.34-hypriotos-v7+ #1 SMP Sun Apr 22 14:57:31 UTC 2018 armv7l

HypriotOS (Debian GNU/Linux 9)

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
_____________________________________________________________________
WARNING! Your environment specifies an invalid locale.
 The unknown environment variables are:
   LC_CTYPE=UTF-8 LC_ALL=
 This can affect your user experience significantly, including the
 ability to manage packages. You may install the locales by running:

   sudo apt-get install language-pack-UTF-8
     or
   sudo locale-gen UTF-8

To see all available language packs, run:
   apt-cache search "^language-pack-[a-z][a-z]$"
To disable this message for all users, run:
   sudo touch /var/lib/cloud/instance/locale-check.skip
_____________________________________________________________________

-bash: /home/client/.cloud-locale-test.skip: Permission denied
client@qw0000p0:~$ ls -lhsa
total 16K
4.0K drwxr-xr-x 4 root   root  4.0K Apr 28 20:57 .
4.0K drwxr-xr-x 3 root   root  4.0K Apr 28 20:57 ..
4.0K drwx------ 2 client users 4.0K Apr 28 20:57 .ssh
4.0K drwxr-xr-x 2 root   root  4.0K Apr 28 20:57 setup
client@qw0000p0:~$
```

I have tried to change the content in the yaml file to be in base64 encoding, in case some invalid characters was present, but it has the same result.

I believe the main problem is that the file that I am writing makes the user (client) not owner of its own home directory! I'm unable to create any files/folders and as can be seen in the `ls -lhsa` output, the file created (`setup/data.json`) is owned by root!

I don't really understand what I am doing wrong here. What is the correct way to write JSON files using hypriot 1.9.0 and cloud-init?

Jacobh2, 2018-07-30:
Tested again but to write the file to `/tmp/data.json` which seems to have worked "better" in the sense that the user's home directory wasn't messed up. 

Now that I ssh into the user I get as expected:
```
> ssh client@192.168.2.2
The authenticity of host '192.168.2.2 (192.168.2.2)' can't be established.
ECDSA key fingerprint is SHA256:8ui/gobzyGFW/2y2tWvHvpOpD4hN7+X+ojcpLCk9V6w.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '192.168.2.2' (ECDSA) to the list of known hosts.
Enter passphrase for key '/.ssh/id_rsa':
Linux qw0000p0 4.14.34-hypriotos-v7+ #1 SMP Sun Apr 22 14:57:31 UTC 2018 armv7l

HypriotOS (Debian GNU/Linux 9)

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
_____________________________________________________________________
WARNING! Your environment specifies an invalid locale.
 The unknown environment variables are:
   LC_CTYPE=UTF-8 LC_ALL=
 This can affect your user experience significantly, including the
 ability to manage packages. You may install the locales by running:

   sudo apt-get install language-pack-UTF-8
     or
   sudo locale-gen UTF-8

To see all available language packs, run:
   apt-cache search "^language-pack-[a-z][a-z]$"
To disable this message for all users, run:
   sudo touch /var/lib/cloud/instance/locale-check.skip
_____________________________________________________________________

HypriotOS/armv7: client@qw0000p0 in ~
$
```
(Also gets the nice colors of the instead of the all back/white when it is not working 🙌 )

If I check the tmp folder, the data is there, but owned by root, so I tried adding
```
owner: "client:users"
permission: "644"
```
to the yaml file when flashing the sdcard. This did not help and checking in the tmp folder:
```
HypriotOS/armv7: client@qw0000p0 in ~
$ ls -lhsa /tmp
total 36K
4.0K drwxrwxrwt  8 root root 4.0K Apr 28 20:57 .
4.0K drwxr-xr-x 21 root root 4.0K Apr 28 20:57 ..
4.0K drwxrwxrwt  2 root root 4.0K Apr 28 20:56 .ICE-unix
4.0K drwxrwxrwt  2 root root 4.0K Apr 28 20:56 .Test-unix
4.0K drwxrwxrwt  2 root root 4.0K Apr 28 20:56 .X11-unix
4.0K drwxrwxrwt  2 root root 4.0K Apr 28 20:56 .XIM-unix
4.0K drwxrwxrwt  2 root root 4.0K Apr 28 20:56 .font-unix
4.0K -rw-r--r--  1 root root  440 Apr 28 20:57 data.json
4.0K drwx------  3 root root 4.0K Apr 28 20:57 systemd-private-0c228bbec790477c9f2481934e550672-systemd-resolved.service-B1El8a

```
It seems like the parameters under `write_files` are not honored? Or am I missing something?

mjjunemann, 2019-03-08:
Having the same problem in hypriot 1.10 is there a solution to this or I'm missing something too?

StefanScherer, 2019-03-09:
The order seems to be write_files before creating users. That‘s how cloud-init works. 

You can try to run commands - I don‘t know if these run after the other both steps.
Try to run chown in the `runcmd` section.