hypriot/flash #135: How to change the keyboard layout?

exocode, 2018-08-01:
Hi there, 

I have a "small" issue:

I dont know how to change the keyboard layout. Can someone point me to the right place?

Thank you!

StefanScherer, 2018-08-01:
I only SSH into my Raspberry Pi's, so I don't know.
But we have added `raspi-config` tool so try this as it's the official helper tool for Raspbian.

StefanScherer, 2018-09-09:
Have you found a solution?
It should be adjustable as with a normal Raspberry Pi SD card image or Debian Linux.

exocode, 2018-10-27:
I wont do that to that through the `raspi-config` tool..
What I try to achieve is, that I flash the HypriotOS with german language already set. At the latest after the first reboot...

I tried a lot of options, no one of them worked, only the terminal language is german now.
But the "Z" is still on the key "Y" and vice versa: Following code shows all my approaches as well the content of the files:

__BTW__: The ONLY WAY I GOT IT WORKING WAS BY USING the `raspi-config`.

**localectl status**
```
System Locale: LAG=de_DE.UTF-8
    VC Keymap: de-latin1nexport
   X11 Layout: DE
    X11 Model: pc105
  X11 Variant: nodeadkeys
```

**/etc/vconsole.conf**
```
KEYMAP=de-latin1-nodeadkeys
```

**cat /etc/locale.conf**
```
LANG=de_DE.UTF-8
export LC_DATE=de_DE.utf8
export LC_NUMERIC=de_DE.utf8
export LC_TIME=de_DE.utf8
export LANG=de_DE.utf8
```

**/etc/default/keyboard**
```
XKBMODEL="pc105"
XKBLAYOUT=de
XKBVARIANT="nodeadkeys"
XKBOPTIONS=""
BACKSPACE="guess"
```

I tried that one: https://wiki.archlinux.de/title/Arch_Linux_auf_Deutsch_stellen#Grundkonfiguration (It didn't work either)

```
ln -s /usr/share/zoneinfo/Europe/Vienna /etc/localtime
```

And that:
- sudo localectl set-x11-keymap de pc105
- sudo localectl --no-convert set-keymap de-latin1-nodeadkeys
- sudo dpkg-reconfigure console-data
- sudo dpkg-reconfigure keyboard-configuration
- sudo localectl --no-convert set-keymap de-latin1-nodeadkeys
- sudo localectl --no-convert set-x11-keymap de pc105 nodeadkeys
- sudo service keyboard-setup restart
- sudo locale-gen de_DE.UTF-8
- sudo locale-gen de
- sudo update-locale de
- sudo update-locale de_DE.UTF-8
- sudo loadkeys de

What else could I try? Best case would be a `cloud-init` with languages already set...

Thank you very much

exocode, 2018-10-28:
I got it working! YAY!

  - sudo apt-get install -y xinput-calibrator
  - sudo apt-get install -y console-data
  - sudo dpkg-reconfigure -f noninteractive keyboard-configuration
  - sudo dpkg-reconfigure -f noninteractive console-data
  - sudo loadkeys de

userdata.yml:

```
packages:
  - software-properties-common
  - python-software-properties
  - xinput-calibrator
  - console-data
  - keyboard-configuration

```

closed 2018-10-28 by exocode

StefanScherer, 2018-10-28:
Thanks for sharing.

bd8392, 2020-11-27:
The raspi-config option is not working anymore...(no keymaps showing) can someone discribe a step by step guide?