hypriot/flash #136: Any chance to use PiBakery instead of flash?

exocode, 2018-08-01:
Hi there I know there is allot of effort to flash and HypriotOS with very small amount of work. 

But is it possible to flash HypriotOS also with the nice Tool PIBakery (master branch/2.0) ? Since 2.0 PiBakery also supports other images than Raspbian. https://github.com/davidferguson/pibakery

It would be nice if someone can give me a hint where to start? HypriotOS uses cloud-init. is it better to adapt something in HypriotOS to use PiBakery? or to write something in PiBakery to flash HypriotOS? 

Thank you in advance for each suggestion...

StefanScherer, 2018-08-01:
No I don't think so. PiBakery looks pretty cool, but we cannot support yet another format.
If you don't want the command line `flash`, then I can recommend https://etcher.io/

exocode, 2018-08-02:
But with Etcher I have not the possibility to "customize". 
Please explain me the basic process how "flash" handles the customization. I will try to write a module for `cloud-init` to support if necessary. Maybe by integrating the flash Tool within PiBakery or removing stuff to make it compatible..

Is `cloud-init` fix integrated into the `HypriotOS` or in the `flash` tool?
Is it enough to remove the `meta-data`, `user-data` and `device-init.yaml` files from the root?
AFAIK `HypriotOS` is based on `Raspbian` and the `raspi-config` is also available, so it seems to be possible...

exocode, 2018-08-02:
PiBakery (may) will start implementing the cloud-init function:

https://github.com/davidferguson/pibakery/issues/161#issuecomment-409874685

closed 2020-01-06 by StefanScherer