hypriot/flash #138: Flash does not set up cloud-init files correctly

ehudkaldor, 2018-08-13:
here is the command I am running:
`→ ./flash --userdata ./home.yml --bootconf ./no-uart-config.txt --device /dev/sdd  ~/Downloads/hypriotos-rpi-v1.9.0.img`

it seems like after all is done, my home.yaml makes it to the /HypriotOS partition as `/user-data`, but it is not applied. at the end of my boot sequence, I see certs generated for the pirate user, and the server is named 'black-pearl', which should have changed.

This is RPI 3b.

MatthiasLohr, 2018-09-11:
I had the same problem. I found that the first line of the yaml file has to be
```
#cloud-config
```

You can also use the cloud-init config validator for checking your config: https://coreos.com/validate/

StefanScherer, 2018-09-22:
Thanks @MatthiasLohr 
I'll check if the ruby YAML gem is installed on macOS automatically, then I can add 

```
ruby -e "require 'yaml';YAML.load_file('./your-user-data-file'); puts 'YAML is ok'"
```

to the flash script, at least on macOS.

humb1t, 2019-10-21:
```
#cloud-config
# vim: syntax=yaml
#

# The current version of cloud-init in the Hypriot rpi-64 is 0.7.9
# When dealing with cloud-init, it is SUPER important to know the version
# I have wasted many hours creating servers to find out the module I was trying to use wasn't in the cloud-init version I had
# Documentation: http://cloudinit.readthedocs.io/en/0.7.9/index.html

# Set your hostname here, the manage_etc_hosts will update the hosts file entries as well
hostname: three_green
manage_etc_hosts: true

# You could modify this for your own user information
users:
  - name: humb1t
    gecos: "Humb1t Pirate"
    sudo: ALL=(ALL) NOPASSWD:ALL
    shell: /bin/bash
    groups: users,docker,video
    plain_text_passwd: ***
    lock_passwd: false
    ssh_pwauth: true
    chpasswd: { expire: false }

# # Set the locale of the system
# locale: "en_US.UTF-8"

# # Set the timezone
# # Value of 'timezone' must exist in /usr/share/zoneinfo
# timezone: "America/Los_Angeles"

# # Update apt packages on first boot
# package_update: true
# package_upgrade: true
# package_reboot_if_required: true
package_upgrade: false

# # Install any additional apt packages you need here
# packages:
#  - ntp

# # WiFi connect to HotSpot
# To make wifi work with RPi3 and RPi0
# you also have to set "enable_uart=0" in config.txt
# See no-uart-config.txt for an example.
#
# # - use `wpa_passphrase SSID PASSWORD` to encrypt the psk
write_files:
  - content: |
      allow-hotplug wlan0
      iface wlan0 inet dhcp
      wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
      iface default inet dhcp
    path: /etc/network/interfaces.d/wlan0
  - content: |
      country=RU
      ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
      update_config=1
      network={
      ssid="***"
      psk="***"
      proto=RSN
      key_mgmt=WPA-PSK
      pairwise=CCMP
      auth_alg=OPEN
      }
    path: /etc/wpa_supplicant/wpa_supplicant.conf

# These commands will be ran once on first boot only
runcmd:
  # Pickup the hostname changes
  - 'systemctl restart avahi-daemon'

  # Activate WiFi interface
  - 'ifup wlan0'
```

Config has no errors in terms of hostname, but device still has black-pearl hostname, user and password works correctly