hypriot/flash #141: Clarify usage without image

Khartir, 2019-01-26:
Both the readme and the usage state:

> If no image is specified, the script will try to configure an existing image. This is useful to try several configuration without the need to rewrite the image every time.

However it is not at all clear how this works. Running `./flash -u user-data` only prints the usage. I assume that this only works with specific options? If so, it would be nice if these are mentioned.

StefanScherer, 2019-03-03:
Oh I think user-data file is interpreted only once on first boot. With occi and device-init it was possible to reconfigure hostname etc.

hannonq, 2019-07-17:
@StefanScherer does that mean that it's no longer possible to make changes without rewriting the images?

Or can I just copy the files manually to /boot/device-init.yaml?

StefanScherer, 2019-07-18:
@hannonq We no longer install device-init, so copying the yaml to /boot doesn’t help.

manicmonkey, 2019-09-28:
Due to the way cloud-init works I think there are some 'instance' related files that would need to be removed from the card to retrigger some of the per-instance modules (such as write_files).

Looks like the 'just update the config' functionality was removed quite some time ago: https://github.com/hypriot/flash/commit/19552a2e7a6e9e277aeb2e88edd24d3ec96a6ead

PR #163 (2020-01-07): Update README
- Remove text about flashing without an image, doesn't work with cloud-init
- Add text about fake-hwclock.data file.
- Update to flash 2.5.0
- Update to HypriotOS 1.12.0

fixes #141
closed 2020-01-07 by StefanScherer