hypriot/flash #148: pv can't show progress

sonnyhcl, 2019-04-29:
https://github.com/hypriot/flash/blob/0b28b5cb525ad27707d4eb57cb54f98b9e3ccdb5/flash#L632

In my use case, the progress soon gets 100% but the dd command is still working at 10MB/s. It seems the image is preloaded in memory.
```
clhu@t5 ~/w/f/sample> flash --userdata static.yml -d /dev/sdd ~/Downloads/installer/hypriotos-rpi-v1.10.0.img 

Is /dev/sdd correct? y
Unmounting /dev/sdd ...
Flashing /home/clhu/Downloads/installer/hypriotos-rpi-v1.10.0.img to /dev/sdd ...
1000MiB 0:00:02 [ 400MiB/s] [=========================================================>] 100%
```

sonnyhcl, 2019-04-29:
Actually, `dd` is now providing a `progress` flag to tell you the progress.
```
status=LEVEL
              The LEVEL of information to print to stderr; 'none' suppresses
              everything but error messages, 'noxfer' suppresses the final
              transfer statistics, 'progress' shows periodic transfer
              statistics
```
https://github.com/hypriot/flash/blob/0b28b5cb525ad27707d4eb57cb54f98b9e3ccdb5/flash#L628-L637
It's time to change this using `progress` flag
```console
echo "Flashing ${image} to ${rawdisk} ..." 
sudo dd bs=$bs_size status=progress "if=${image}" "of=${rawdisk}" 
```

shafer, 2020-08-10:
I don't think it is `pv` that is the problem. I think the problem is that `dd` needs flags to bypass cache(s) so that writes don't appear to finish super fast and you see the real progress. I added `oflag=direct iflag=fullblock` to my dd options and then the progress looks correct. I don't know if macOS supports those options. If there is interest I'm game to make a PR that adds support to the script so that the progress shown by pv reflects reality.