hypriot/flash #156: User-data "docker swarm join --token..." not taken into account at boot

sountrust, 2019-10-17:
Hello I'm a newbie on docker swarm and user-data file
I tried to add a new node to a cluster directly at boot setup with "docker swarm join --token " command in user-data file with no avail
Here is a sample of the file:
[image]
I successfully ran the command with the same token directly into the node though
The purpose is to create a whole raspberry cluster with all workers and managers by configuring data-user file into each Hypriotos thus directly set up with hostnames, services, workers, managers etc... at boot stage.
Whether I'm totally wrong or missing something I would be glad hearing advices from any of you 
Thank you

sountrust, 2019-11-03:
Any thoughts ?

StefanScherer, 2019-11-04:
These runcmd may be running too eatly on boot. The fake clock/ntp might not have correct time at that point. 
I had a complex script to create a swarm after powering on the tiny cloud. It recreated the whole swarm due to the missing battery backed hardware clock.