hypriot/flash #157: how to change hostname with flash hypriotos-rpi-v1.11.4.img.zip

xiaods, 2019-11-10:
i follow the documents and flash arge helps , and successful flash the hypriotos-rpi-v1.11.4.img.zip to SD.  but i have try many time without successful. the hypriotos already init with default hostname
black-pearl.local. anyone can give some hints for me. thanks a lot.

xiaods, 2019-11-10:
when airgap install failed, i reinstall by one-installation script:
```
url -sfL https://get.k3s.io | sh -
```
it always failed by tls handshake failed.
dubug logs use journalctl -xe , see blow messages:
```
cycle,LimitRanger,ServiceAccount,NodeRestriction,TaintNodesByCondition,Priority,DefaultTolerationSeconds,DefaultStorageClass,StorageObjectInUseProtection,MutatingAdmissionWebhook,RuntimeClass.
Nov 10 03:20:32 black-pearl k3s[1974]: I1110 03:20:32.572830    1974 plugins.go:161] Loaded 7 validating admission controller(s) successfully in the following order: LimitRanger,ServiceAccount,Priority,PersistentVolumeClaimResize,ValidatingAdmissionWebhook,RuntimeClass,ResourceQuota.
Nov 10 03:20:32 black-pearl k3s[1974]: time="2019-11-10T03:20:32.823956070Z" level=info msg="Running kube-scheduler --bind-address=127.0.0.1 --kubeconfig=/var/lib/rancher/k3s/server/cred/scheduler.kubeconfig --leader-elect=false --port=10251 --secure-port=0"
Nov 10 03:20:32 black-pearl k3s[1974]: time="2019-11-10T03:20:32.905733232Z" level=info msg="Running kube-controller-manager --allocate-node-cidrs=true --bind-address=127.0.0.1 --cluster-cidr=10.42.0.0/16 --cluster-signing-cert-file=/var/lib/rancher/k3s/server/tls/server-ca.crt --cluster-signing-key-file=/var/lib/rancher/k3s/server/tls/server-ca.key --kubeconfig=/var/lib/rancher/k3s/server/cred/controller.kubeconfig --leader-elect=false --port=10252 --root-ca-file=/var/lib/rancher/k3s/server/tls/server-ca.crt --secure-port=0 --service-account-private-key-file=/var/lib/rancher/k3s/server/tls/service.key --use-service-account-credentials=true"
Nov 10 03:20:33 black-pearl k3s[1974]: I1110 03:20:33.577549    1974 controllermanager.go:161] Version: v1.16.2-k3s.1
Nov 10 03:20:33 black-pearl k3s[1974]: I1110 03:20:33.595590    1974 deprecated_insecure_serving.go:53] Serving insecurely on [::]:10252
Nov 10 03:20:33 black-pearl k3s[1974]: I1110 03:20:33.632742    1974 server.go:143] Version: v1.16.2-k3s.1
Nov 10 03:20:33 black-pearl k3s[1974]: I1110 03:20:33.636746    1974 defaults.go:91] TaintNodesByCondition is enabled, PodToleratesNodeTaints predicate is mandatory
Nov 10 03:20:33 black-pearl k3s[1974]: W1110 03:20:33.690567    1974 authorization.go:47] Authorization is disabled
Nov 10 03:20:33 black-pearl k3s[1974]: W1110 03:20:33.693645    1974 authentication.go:79] Authentication is disabled
Nov 10 03:20:33 black-pearl k3s[1974]: I1110 03:20:33.697008    1974 deprecated_insecure_serving.go:51] Serving healthz insecurely on [::]:10251
Nov 10 03:20:43 black-pearl k3s[1974]: time="2019-11-10T03:20:43.234004978Z" level=fatal msg="starting tls server: Get https://127.0.0.1:6444/apis/apiextensions.k8s.io/v1beta1/customresourcedefinitions: net/http: TLS handshake timeout"
Nov 10 03:20:43 black-pearl systemd[1]: k3s.service: Main process exited, code=exited, status=1/FAILURE
```

StefanScherer, 2019-11-10:
We have moved from device-init to cloud-init some time ago. The option in flash is obly there to support old sd card images. Maybe it‘s time to remove this option. 
Please have a look at the samples https://github.com/hypriot/flash/blob/master/sample/README.md

xiaods, 2019-11-10:
@StefanScherer thanks a lot. it resolve my case.

closed 2019-11-10 by xiaods