hypriot/flash #158: Support multiple user-data?

mrmillers, 2019-11-13:
According to cloud-init, multiple user data file could be supplied.
I was trying to do something like this:

flash -d /dev/sda \
--userdata user-data.yml \ # user config
--userdata os-init.sh \ # a script to run on start up
image.img

Only last file is copied.

mrmillers, 2019-11-13:
A walk around is create the script with write_files section and invoke it with runcmd section. But this is a little bit inconvenient.

firecyberice, 2019-12-20:
Currently there is a [flag](https://github.com/hypriot/flash/blob/master/flash#L41) to copy additional files to the boot partition. This file can then be invoked in the runcmd section, but you do not need to use write_files for that.

But PRs to support multiple user-data files are also welcome. :smiley: 
Feel free to reopen this issue.

closed 2019-12-20 by firecyberice