hypriot/flash #159: --hostname only works if --userdata already contains a hostname section

tommack, 2019-11-27:
At the very least a warning would be helpful if `--hostname` is used but there is nothing to replace.

rafaeleyng, 2020-06-01:
Could you elaborate? I had what seems to be the exact opposite problem. I've tried to set the hostname via `--userdata` to no avail. But `--hostname` worked just fine.

killerwhile, 2021-01-19:
A bit more details: in image such as ubuntu 20.04, which doesn't include `user-data` file by default, if no `user-data` file is provided or the one provided doesn't include a `hostname: xxx` section, the parameter `--hostname` doesn't do anything.
flash is testing the presence of user-data file and replace an existing `hostname: xxx` directive to what is provided.