hypriot/flash #160: WSL support?

testautomation, 2019-12-16:
Tried running in WSL and got the message `This script does not work in WSL.` 
What hinders flash to support WSL? WSL matured a lot in the last years. May be it's worth to give it another chance? :)

StefanScherer, 2019-12-17:
Nice try 🤗
From what I know on Windows is that we need real admin access to USB devices to write a SD card.
And WSL only runs in normal user context, so this might be the hard problem here.

Actually, the check in `flash` doesn't work in WSL2, but it doesn't list me any USB devices.
I'll add a "fix" to give out a message that WSL2 is also not supported.

testautomation, 2019-12-18:
Oh, ok. I was not aware of this.

Since WSL / WSL2 I thought it might be worth to give Windows another chance as my main OS. Seems that its a waste of time and this case is another of endless examples which proves me right 😄

closed 2019-12-18 by testautomation