hypriot/flash #164: unable to pull docker images on first boot

MaartenKleijwegt, 2020-01-09:
When I flash a raspberry pi zero w it seems to be unable to run docker containers from the runcmd part of the user-data

First of all, it throws the following error:

> wpa_supplicant: /sbin/wpa_supplicant daemon failed to start
> run-parts: /etc/network/if-pre-up.d/wpasupplicant exited with return code 1
> Failed to bring up wlan0.

However, it is connected to my wifi as I can see in my router. Moreover, it was able to fetch my github ssh keys and performs apt update fine (I've put this in runcmd to make sure it is done after wifi has been connected)

However, after that I try to pull the portainer image but then it throws an error:

> error response from daemon get https //registry-1.docker.io/v2/: dial tcp: lookup registry-1.docker.io on [::1]:53: read udp [::1]:33910->[::1]:53: read: connection refused 

I know wifi isn't officially supported by cloud-init but I'm not sure it's a wifi related issue. All other internet related things seem to work. Also the error message seems to suggest it wasn't unable to connect, but the connection was refused somehow.

When I ssh into the device the docker run command works as normal. Also when i try the rainbow example (which uses docker swarm) everything seems to work fine

My user-data.yml for reference:

#cloud-config
# vim: syntax=yaml
#

# The current version of cloud-init in the Hypriot rpi-64 is 0.7.9
# When dealing with cloud-init, it is SUPER important to know the version
# I have wasted many hours creating servers to find out the module I was trying to use wasn't in the cloud-init version I had
# Documentation: http://cloudinit.readthedocs.io/en/0.7.9/index.html

# Set your hostname here, the manage_etc_hosts will update the hosts file entries as well
hostname: roger
manage_etc_hosts: true

# You could modify this for your own user information
users:
  - name: someusername               # use any user name you like
    primary-group: users
    shell: /bin/bash
    sudo: ALL=(ALL) NOPASSWD:ALL
    groups: users,docker,adm,dialout,audio,plugdev,netdev,video
    lock_passwd: false
    ssh-import-id: gh:somegithubid
  - name: pirate
    gecos: "Hypriot Pirate"
    sudo: ALL=(ALL) NOPASSWD:ALL
    shell: /bin/bash
    groups: users,docker,video
    plain_text_passwd: hypriot
    lock_passwd: false
    ssh_pwauth: true
    chpasswd: { expire: false }

# # Set the locale of the system
locale: "en_US.UTF-8"

# # Set the timezone
# # Value of 'timezone' must exist in /usr/share/zoneinfo
timezone: "Europe/Amsterdam"
# # Update apt packages on first boot
# package_update: true
# package_upgrade: true
# package_reboot_if_required: true
package_upgrade: false

# # Install any additional apt packages you need here
# packages:
#  - ntp
# # WiFi connect to HotSpot
# To make wifi work with RPi3 and RPi0
# you also have to set "enable_uart=0" in config.txt
# See no-uart-config.txt for an example.
#
# # - use `wpa_passphrase SSID PASSWORD` to encrypt the psk
write_files:
  - content: |
      allow-hotplug wlan0
      iface wlan0 inet dhcp
      wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
      iface default inet dhcp
    path: /etc/network/interfaces.d/wlan0
  - content: |
      country=NL
      ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
      update_config=1
      network={
      ssid="somessid"
      psk="somepassword"
      proto=RSN
      key_mgmt=WPA-PSK
      pairwise=CCMP
      auth_alg=OPEN
      }
    path: /etc/wpa_supplicant/wpa_supplicant.conf
# These commands will be ran once on first boot only
runcmd:
  # Pickup the hostname changes
  - 'systemctl restart avahi-daemon'

  # Activate WiFi interface
  - 'ifup wlan0'

  # sleep to wait for wifi to get up
  - 'sleep 10'

  # Update apt packages on first boot
  - 'apt-get update'
  - 'apt-get upgrade'
  
  # startup my docker containers
  - [ sh, -c, echo "Starting Portainer now..." ]
  - [ sh, -c, 'docker volume create portainer_data' ]
  - [ sh, -c, 'docker run -d -p 9000:9000 -p 8000:8000 --name portainer --restart always -v /var/run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer' ]
  - [ sh, -c, echo "Finished starting Portainer" ]

geoff-coppertop, 2020-05-11:
I see a similar issue when I use an ethernet connection on an RPi3

MaartenKleijwegt, 2020-05-12:
I've seen it too on ethernet connection. I found out it had to do something with the user, or lack there of, that's issuing the commands.

As mentioned before using docker swarm does work since it starts the containers differently.

Maybe a command can be sent to swap user and it will work. Conclusion is I gave up on hypriot entirely as the software didn't match my use-case that well.

flochtililoch, 2020-08-30:
Related: https://github.com/hypriot/image-builder-rpi/issues/342