hypriot/flash #166: HyperiotOS very unstable with Raspberry Pi 4

djermann, 2020-01-31:
I have set up a Raspberry Pi 4 with the flash tool with HyperiotOS v1.12.0 and the default user-data cloud-init file but when I plug it in and start it up it sometimes does not start at all, sometimes it starts but the root filesystem is mounted read-only, sometimes the /boot partition is not mounted and sometimes everything just works, but after a reboot it suddenly does not work anymore. I tested this on two different Raspberry Pi 4 with two different SD-cards with the same results. It behaves like this with a fresh install and gets worse as I change some partitions.

OS: HyperiotOS v1.12.0
Hardware: Raspberry Pi 4 Model B (4 GB RAM) with power over ethernet hat, 128 GB Samsung microSD card EVO Plus.
Flash command used: `flash hypriotos-rpi-v1.12.0.img`
Partitions:
```
Device     Boot    Start       End   Sectors   Size Id Type
/dev/sdb1  *        2048    133119    131072    64M  c W95 FAT32 (LBA)
/dev/sdb2         133120   6424575   6291456     3G 83 Linux
/dev/sdb3        6424576  23201791  16777216     8G 82 Linux swap / Solaris
/dev/sdb4       23201792 250085375 226883584 108.2G  5 Extended
/dev/sdb5       23203840  29495295   6291456     3G 83 Linux
/dev/sdb6       29497344 250085375 220588032 105.2G 83 Linux

```

/etc/fstab: 
```
proc /proc proc defaults 0 0
PARTUUID=572083f2-01 /boot vfat defaults 0 0
PARTUUID=572083f2-02 / ext4 defaults,noatime 0 1
PARTUUID=572083f2-03 none swap sw 0 0
PARTUUID=572083f2-05 /tmp ext4 defaults,noatime 0 2
PARTUUID=572083f2-06 /opt ext4 defaults,noatime 0 2
```
I have not changed anything else.

StefanScherer, 2020-05-21:
@djermann can you retry with the latest release v1.12.1? We've fixed an issue at boot time.

djermann, 2020-05-22:
@StefanScherer I tried again with the latest release and was not able to reproduce this problem. So far it looks good. Thank you very much for resolving this!

closed 2020-05-22 by djermann

StefanScherer, 2020-05-22:
Thanks for testing again and the feedback. Enjoy!