hypriot/flash #169: Flash error: hdiutil: couldn't unmount "disk2" - Resource busy

xUnholy, 2020-03-18:
Whilst running my flash command as follows;

```flash \
    --userdata setup/cloud-config.yml \
    ~/Downloads/ubuntu-19.10.1-preinstalled-server-arm64+raspi3.img
```

I get a sporatic error where my flash gets to the end and then fails with the following response: 
```
Is /dev/disk2 correct? y
Unmounting /dev/disk2 ...
Unmount of all volumes on disk2 was successful
Unmount of all volumes on disk2 was successful
Flashing ~/Downloads/ubuntu-19.10.1-preinstalled-server-arm64+raspi3.img to /dev/rdisk2 ...
Password:
0+47533 records in
0+47533 records out
3115115520 bytes transferred in 109.241900 secs (28515757 bytes/sec)
Mounting Disk
Mounting /dev/disk2 to customize...
Copying cloud-init setup/cloud-config.yml to /Volumes/system-boot/user-data ...
Unmounting /dev/disk2 ...
hdiutil: couldn't unmount "disk2" - Resource busy
Something went wrong.```

xUnholy, 2020-03-18:
Fixing this locally on my MacOS I just tweaked the following:

```
echo "Unmounting ${disk} ..."
sleep 5
```

I'm guessing there might have been some type of race condition or the copy operation wasn't completed and/or verified before performing the detach so increasing the sleep to 5 was adequate for the copy to complete gracefully.

StefanScherer, 2020-04-01:
Closed by #171

closed 2020-04-01 by StefanScherer

xUnholy, 2020-04-01:
Thanks @StefanScherer - look forward to the next release 👍

StefanScherer, 2020-04-01:
@xUnholy flash version 2.6.0 is released. :shipit: