hypriot/flash #173: maintain a homebrew package for flash

onedr0p, 2020-04-14:
The install instructions are pretty easy, but it would be nice to have a homebrew  package for updating.

onedr0p, 2020-04-14:
If it's helpful, this is how it is done for kubetail which is also a bash script

https://github.com/johanhaleby/homebrew-kubetail

StefanScherer, 2020-04-14:
Thanks for the suggestion. 
There is a pull request https://github.com/Homebrew/homebrew-core/pull/52503 which you can follow.

witsch, 2020-04-14:
Unfortunately, getting the pull request (or rather the required test) to be in an acceptable form turned out to be a bit tricky.  Now it should be okay, but it might take another bit for the maintainers to get back to it… 🙂

onedr0p, 2020-04-14:
No worries, the homebrew maintainers are super stingy on most everything. There's always the way I listed above if they don't approve it. :)

StefanScherer, 2020-04-16:
It just landed, thanks @witsch !!! 🎉🎉  I couldn't believe this was possible 😅 

[image]

closed 2020-04-16 by StefanScherer

witsch, 2020-04-16:
You're welcome, I'm glad I could contribute a little bit… and hey, now it's only [one step](https://github.com/witsch/stacks/commit/385ec91f) to get a new Pi running!  Well, almost… 🙂

StefanScherer, 2020-06-01:
Ping @witsch I don't remember if the homebrew `flash` formula gets updated automatically. I just released version 2.7.1 https://github.com/hypriot/flash/releases/tag/2.7.1 🤗

witsch, 2020-06-01:
Pong 🙂

They're not, but updating has become very convenient by now. So in this case — and this is mostly meant as a 'note to self' — all that's needed to create https://github.com/Homebrew/homebrew-core/pull/55634 was:
```bash
$ brew bump-formula-pr --url https://github.com/hypriot/flash/releases/download/2.7.1/flash --sha256 …
```

StefanScherer, 2020-06-02:
Thank you @witsch !
This already creates the PR in homebrew-core?

Do you know if I can run this on Linux here in our pipeline? -> https://github.com/hypriot/flash/blob/master/.circleci/config.yml#L55

witsch, 2020-06-02:
Yes, it does. I'm not sure how easily it integrates in CI, though. It needs your Github credentials, of course, but then again perhaps all you need is a token in the environment? Would you let me know if you get it to work, please? 🙂