hypriot/flash #174: MacOS dd: invalid number: ‘1m’

xUnholy, 2020-04-16:
The following causes an issue on a fresh MAC running Mojave v10.14.6

https://github.com/hypriot/flash/blob/aa8107c03b96ae31544850bf269c9e214743fbf3/flash#L146

I fixed this locally by just making it as follows:

```
    bs_size=1M
```

@StefanScherer happy to raise a PR to fix this? Also the latest v2.7.0 did not work on my Mac at all. I'll look into that next, as I downgraded back to v2.6.0.

The following was the issue on v2.7.0 with MacOS

```
Is /dev/disk2 correct? y
Unmounting /dev/disk2 ...
Unmount of disk2 failed: at least one volume could not be unmounted
Unmount was dissented by PID 24954
Unmount of disk2 failed: at least one volume could not be unmounted
Unmount was dissented by PID 2495
```

StefanScherer, 2020-04-17:
Thanks @xUnholy for the report. 
Can you give me a little more detail about your machine?
What does
```
which dd
ls -l $(which dd)
```
show you?

For me it's 
```
$ which dd
/bin/dd
$ ls -l $(which dd)
-rwxr-xr-x  1 root  wheel  45120 Mar 17 16:42 /bin/dd
```
I am on macOS Catalina 10.15.4, but also had a machine with 10.14.x for a long time without a problem.

We had such reports from time to time, eg. #131 but it turned out that `dd` was not the original one.

searsaw, 2020-04-25:
Have this same issue. Mine points to the version brought in by `coreutils` from Homebrew.

```
/usr/local/opt/coreutils/libexec/gnubin/dd -> ../../bin/gdd
```

I "fixed" it by putting `/bin` at the beginning of my PATH again.

```
export PATH="/bin:$PATH"
```