hypriot/flash #176: Does the WiFi example really (still) work?

marcelstoer, 2020-05-12:
I haven't had any success to enable WiFi with the latest HypriotOS on a Raspberry Pi 4. My process is as follows (for all HypriotOS installations I do) on macOS:

- download latest img/zip
- mount img
- edit user-data
- flash the img with Etcher

That has worked well so far.

However, if I enable WiFi I never manage to SSH into the Pi. Since I currently don't have a micro HDMI cable I can't even analyze where it's stuck.

I also checked and tested the WiFi example at https://github.com/hypriot/flash/blob/master/sample/wifi-user-data.yml but it looks identical to the blueprint included in the latest image.

The Pi / router / WiFi etc. are fine. I verified by configuring and flashing Raspian and all is well.

Macf41, 2020-05-16:
Same problem here. I used the latest image

StefanScherer, 2020-05-21:
@marcelstoer yes it still should work. Can you try with the latest release v1.12.1 again? We've fixed an issue at boot that may be related.

mattvonrocketstein, 2020-06-19:
I'm having issues with this for v1.12.2 as well

Sijoma, 2020-12-05:
I also cant seem to get it to work. I'm using the latest relase (v1.12.3)

alexwhittemore, 2020-12-08:
Same problem (maybe) - using the latest 1.12.3. If I use the --password mywifipass --ssid mynetworkname (or -p/-s), apparently nothing happens. flash makes no indication it processed those inputs at all:

(note: changed the binary name to hypriot_flash so I have a clue what it is years from now when I wonder)

```
-> % hypriot_flash -n owl-hypriot-1 --ssid SSID --password PASSWORD ./hypriotos-rpi-v1.12.3.img.zip 
/usr/bin/unzip
Uncompressing ./hypriotos-rpi-v1.12.3.img.zip ...
Archive:  ./hypriotos-rpi-v1.12.3.img.zip
  inflating: /tmp/hypriotos-rpi-v1.12.3.img  
Use /tmp/hypriotos-rpi-v1.12.3.img
No SD card found. Please insert SD card, I'll wait for it...

Is /dev/disk2 correct? 
Please answer yes or no.

Is /dev/disk2 correct? y
Unmounting /dev/disk2 ...
Unmount of all volumes on disk2 was successful
Unmount of all volumes on disk2 was successful
Flashing /tmp/hypriotos-rpi-v1.12.3.img to /dev/rdisk2 ...
Password:
1.27GiB 0:00:31 [41.1MiB/s] [===================================================================================>] 100%            
0+20800 records in
0+20800 records out
1363148800 bytes transferred in 25.937416 secs (52555305 bytes/sec)
Mounting Disk
Mounting /dev/disk2 to customize...
Set hostname=owl-hypriot-1
Unmounting /dev/disk2 ...
"disk2" ejected.
Finished.
```
Additionally, if I mount the SD and look at `/Volumes/HypriotOS/user-data`, the whole wifi section remains commented out.

if I use the example userdata @marcelstoer mentioned (thanks!) and `hypriot_flash -n owl-hypriot-1 -u wifi-user-data.yml  ./hypriotos-rpi-v1.12.3.img.zip` it definitely does get copied to the image. But my pi apparently never connects, and it looks like serial output is disabled so I can't tell why not or what it's trying to do.

mjkl-gh, 2021-01-19:
I know how you feel. Thusfar I've not come far with this

when reading this pull request:

https://github.com/hypriot/flash/pull/165

it mentions still requiring an uncommented user-data file

If you look inside flash (it's a bash script, not a binary)
https://github.com/hypriot/flash/blob/c501e624f6361851ac76bc810ebb464bca88d411/flash#L754

it replaces the ssid and passwd lines. However, if these are still commented (per default) it either won't match or won't uncomment I guess?

I even didn't have the luck of a working wifi-user-data.yml file. it does do some of the commands specified. But both files written by write_files do not exist. Even worse something seems broken about the wlan0 config as sudo raspi-config errors out saying no communication possible with wpa_supplicant. wpa_supplicant is running though.

mattvonrocketstein, 2021-01-19:
IDK about other people, but this exact feature of very simple and pre-configured wifi is a big part of why I was using hypriot tools for headless stuff originally.  i think there's a lot of reasons to no longer prefer this approach though.  

as far as i know newer Ubuntu versions a) support netplan, b) support cloud-init, and c) have no issues installing or running docker on most raspberry pi versions but your milage may vary for the super old hardware releases.  given this set of features, i'm not sure what the use-case is now for hypriot or even raspbian.