hypriot/flash #177: Can't make HypriotOS 1.12.1 work. So many different problems

rafaeleyng, 2020-06-01:
TLDR:

- I've spent 2 full days trying to use HypriotOS. I've had **so many problems** that I can't give a summary, so please read the full issue.
- After all of this, I'm **still unable to run HypriotOS**.
- I've encountered a lot of **missing, outdated, obscure, and contradictory documentation**. Please read carefully the full issue where I either represent these points in one of three ways: as questions, by explicitly saying that the doc could explain the specific point, or making guesses how something should work (instead of getting clear information from the docs).
- I would **really like to be able to run this** on my Raspberry Pi devices since it seems to be a perfect fit for what I need to do.
- I'm **willing to help to improve the documentation** if you think my observations are valid. But I can only do so **after understanding the enormous confusion that I'm facing** with this setup. I've put around 6 hours of work just to write this issue with every detail I could get.
- I would like to make it very explicit (with all due respect for the project): using HypriotOS and the flash script was **not easy at all** in my experience (unlike what is said [here](https://blog.hypriot.com/faq/#is-there-an-easy-way-to-flash-the-image-of-hypriotos-onto-a-sd-card)).

---

Hello there

First and foremost, thank you for the project, it seems to really provide a good solution for the problem, especially for less experienced developers like me. If the following text ends up sounding harsh, it is not directed to the project, or to the developers, **but to my own frustration**.

It's been a recurring theme for me. Whatever I try to learn or do, I end up finding a tool that "makes everything really easy". More often than not, I end up not being able to use or understand the tool, and left with the doubt: either it isn't that easy, or I'm just very, very incapable.

I've spent the 2 last days trying to use Hypriot, and failing miserably. I'm so confused about how this is supposed to work that I can't even open a specific issue about a specific question, so I'll try to expose the problems I've had in this single issue and we can work from this.

Hopefully my questions could point some obscure points about the documentation, and so help more people who are in my situation.

Note: I've read several materials before creating this issue. To cite some:
- https://blog.hypriot.com/post/releasing-HypriotOS-1-11/
- https://blog.hypriot.com/getting-started-with-docker-on-your-arm-device/
- https://blog.hypriot.com/getting-started-with-docker-and-mac-on-the-raspberry-pi/
- https://blog.hypriot.com/faq/#wifi
- https://www.raspberrypi.org/documentation/configuration/config-txt/README.md
- https://cloudinit.readthedocs.io/en/0.7.9/topics/modules.html
- I've cloned, read all the docs, and looked at all samples from https://github.com/hypriot/flash
- I've cloned, and skimmed through https://github.com/hypriot/image-builder-rpi

I hope that makes it clear that this is not just a lazy "help me, it is not working" without any effort or context.

---

# what devices I have

- Mac Book (where I'm flashing the SD cards etc)
- Raspberry Pi 2 Model B (with a wi-fi stick)
- Raspberry Pi Zero W

---

# what I'm trying to achieve

I want to flash an image on my devices with HypriotOS v1.12.1, configuring:
- hostname
- wifi settings
- SSH keys

To make it more clear: I want to put an SD card in my Raspberry Pi devices and boot them with these 3 things configured, so I don't want to have to plug a keyboard or Ethernet cable to my devices.

---

# what I've tried

## with the Raspberry Pi 2 Model B

First I've tried a bunch of stuff that I can't remember right now. Then I started all over again, with baby steps (doing the most basic setup, then trying to change one thing at a time), as follows.

### first attempt - no config

Here I'm doing basically what is described [here](https://blog.hypriot.com/post/releasing-HypriotOS-1-11/#quick-start).

The blog post is about HypriotOS 1.11 (I'm actually using 1.12). Not sure whether that makes any difference here.

Steps:
- flash the image without any extra configuration or file, using the `flash` script: `flash -f ~/cluster-images/hypriotos-rpi-v1.12.1.img`
- connect the Pi via Ethernet to my Ethernet switch, via cable
- access the Pi via SSH with the default hostname: `ssh pirate@black-pearl.local`, and password `hypriot`.
[... 209 lines cut ...]
[image]

Here another inconsistency in the doc emerges:
- https://blog.hypriot.com/faq/#how-can-i-boot-a-raspberry-pi-zero implies that the `--userdata` is all we need to make wifi work on a Pi Zero
- https://github.com/hypriot/flash says:
  > The options --userdata and --bootconf must be used to disable UART and enable onboard WiFi for Raspberry Pi 3 and Pi 0. For external WiFi sticks you do not need to specify the -bootconf option.

  So I understand that different that explained in the FAQ, we do actually need to pass `--bootconf` with a file containing `enable_uart=0` (just like the example https://github.com/hypriot/flash/blob/master/sample/no-uart-config.txt). Besides that, I could not understand *at all* the relation between UART and wifi, and the linked resource (https://www.raspberrypi.org/documentation/configuration/config-txt/README.md, which links to https://www.raspberrypi.org/documentation/configuration/device-tree.md#part4.6, which links to https://www.raspberrypi.org/documentation/configuration/uart.md) does not explain it at all. So I did this blindly following orders, and not seeing any result.

Long story short, whether I configured or not the UART thing, wifi didn't work on my Raspberry Pi Zero, with the error shown in the picture above

---

# other observations about the docs

I'll just list some other points where I've found the docs confusing.

- https://blog.hypriot.com/getting-started-with-docker-on-your-arm-device/ links to https://blog.hypriot.com/getting-started-with-docker-and-mac-on-the-raspberry-pi/ which has a step by step on how to flash the image on a Mac Book, but does not use the flash script, which other parts of the doc, like https://blog.hypriot.com/faq/#wifi, seems to prefer.

- the default hostname could be documented at the FAQ (I know by now it is `black-pearl`, this hostname appears in some examples, but is never stated as the default hostname). For contrast, the default credentials are very well documented at https://blog.hypriot.com/faq/#default-credentials.

- https://blog.hypriot.com/faq/#how-can-i-change-the-hostname suggests:
  > Just edit the /boot/user-data file with an editor before you flash the SD image.
  
  How/where do I find this file **before** flashing the SD image? Once I've flashed the image to the SD card, then I know the file is at `/Volumes/HypriotOS/user-data`, but I don't understand how to edit this file before even flashing it to the SD card. The only thing I can imagine is that this file is contained somehow in the image `hypriotos-rpi-v1.12.1.img` I've downloaded. Should I open this image to find the `/boot/user-data` file in there, then edit it and then generate a new image, with this file changed? If so, any instructions on how to do it? I have the impression that the docs mean **after** instead of **before**, but this is just my guess.

StefanScherer, 2020-06-01:
Hey @rafaeleyng Thank you very much for your detailed repot. Oh you really tried hard and the time you put into getting it work and writing this feedback is extraordinary.

I haven't read all scenarios yet as I want to give you some quick tips that might help. But be assured I'll read all and reply with more comments later.

So, the basic tests worked fine, but as soon as you created an own coud-init.yaml it started to fail.

One of the nasty things about cloud-init is that this yaml file must contain a first line containing

```
#cloud-config
```

Thanks to your feedback I realise that this should be made more visible in our docs.
If that line is missing, cloud-init does not recognise this file and then, of course, none of the configurations are applied and won't work.

Maybe this little piece is the only thing that unblocks you.

StefanScherer, 2020-06-01:
Here is a link to the cloud-init FAQ https://cloudinit.readthedocs.io/en/latest/topics/faq.html#how-can-i-debug-my-user-data

> Two of the most common issues with user data, that also happens to be cloud-config is:
> 1. Incorrectly formatted YAML
> 2. First line does not contain #cloud-config

It's hard to add a yaml check into `flash` bash script, but at least we could check if the first line has the comment.

StefanScherer, 2020-06-01:
I just found out that on macOS we already validate the YAML file. So the missing comment makes it to the top 1 problem. I've opened #178 to add a check for the comment in the first line of the user-data file.

YasserAntonio, 2020-06-01:
Hello, I  upgraded my mac to Catalina and updated flash to it last version and I have the exacte same problem described in this issue with my rasp 3 and 4. Before everything was working well since I used the cloud-init config and flash several time before without any issue to report.

rafaeleyng, 2020-06-01:
Hey @StefanScherer, thanks a lot for the quick response. Let's work together and fix this 😄 .

I'll just like to point out that, even if we manage to solve this problem, **the main issue I see here is the obscurity/lack of consistency of the docs**. If the docs were clearer, I might have ended with the same problem, but I sure **wouldn't have to try a hundred combinations of factors and play the guessing game**. And I think my original post helps to point out several of these inconsistencies.

---

> One of the nasty things about cloud-init is that this yaml file must contain a first line containing

I'd take a step back here.

As somebody who doesn't know about `cloud-init` (just found out that it existed 2 days ago), this is one of the most obscure parts of all of this. Which file are we even referring to? `userdata` or `metadata`? What the difference between each of them? What should I put in which one of them? All of this is very implicit, especially in cloud-init doc (which does not include comprehensive documentation of both files, what are all the valid configurations, etc).

[image]

No one of these menus actually explains what it is. The first menu "User-Data Formats" talks a lot about file formats, shows some Python code, I and can't see the relationship between all of that and the `yaml` file we are talking about here.

The second menu "Cloud config examples" seems better, it does look like the `yaml` we are talking about (I still don't see the relationship of this with the User-Data Formats). And then it follows with a long list of examples that should supposedly just work only benefits people who already know what that actually does.

I'll not dive into the "metadata" part, whose documentation is even more obscure and confusing. But my point remains.

Let's talk about the issue in hand now.

---

> ```#cloud-config```

So, unfortunately, that was not my problem. I removed the comments in my examples shown in the original post, but in most of them (not all) I've actually copied examples from:
- https://blog.hypriot.com/faq/#wifi
- https://github.com/hypriot/flash/blob/master/sample/wifi-user-data.yml

in which they all have the `#cloud-config` comment.

---

To be 100% sure, I have just tested it again:
- Raspberry Pi 2 Model B (so, no onboard wifi, but using a wifi stick). Still not sure whether this makes any difference (see my main post for all the confusion about needing an extra step for devices with onboard wifi)
- this exact configuration for `--userdata` (except for the password):
  ```yml
  #cloud-config

  # Set your hostname here, the manage_etc_hosts will update the hosts file entries as well
  hostname: black-pearl
  manage_etc_hosts: true

  # You could modify this for your own user information
  users:
    - name: pirate
      gecos: "Hypriot Pirate"
      sudo: ALL=(ALL) NOPASSWD:ALL
      shell: /bin/bash
      groups: users,docker,video
      plain_text_passwd: hypriot
      lock_passwd: false
      ssh_pwauth: true
      chpasswd: { expire: false }

  package_upgrade: false

  # # WiFi connect to HotSpot
  # # - use `wpa_passphrase SSID PASSWORD` to encrypt the psk
  write_files:
    - content: |
        allow-hotplug wlan0
        iface wlan0 inet dhcp
        wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
        iface default inet dhcp
      path: /etc/network/interfaces.d/wlan0
    - content: |
        country=BR
        ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
        update_config=1
        network={
        ssid="TP-Link_11FE_5G"
        psk="--------"
        proto=RSN
        key_mgmt=WPA-PSK
        pairwise=CCMP
        auth_alg=OPEN
        }
      path: /etc/wpa_supplicant/wpa_supplicant.conf

  # These commands will be ran once on first boot only
  runcmd:
    # Pickup the hostname changes
    - 'systemctl restart avahi-daemon'

    # Activate WiFi interface
    - 'ifup wlan0'
  ```

Some pictures of what I've got (note that these 3 pictures are from the Raspberry Pi 2 Model B, while in my original post they were all about the Raspberry Pi Zero, I don't have any pictures of the initial tests with the Pi 2 to compare).

Networking service:
[image]

ifconfig:
[image]

wpa supplicant:
[image]

Then I've retested the same configuration with the Pi Zero, and my results were the same than in the original post (including the pictures).

---

@YasserAntonio has a really good point. I've had several problems with several applications after upgrading OS X to Catalina.

rafaeleyng, 2020-06-01:
Might be related: https://github.com/balena-io/etcher/issues/2997

rafaeleyng, 2020-06-01:
I just finished trying to set up the Raspberry Pi 2 Model B again, but with the Vagrant setup, in a Linux VM. I was able to write the image to the card and added `--userdata` with the same configuration than I used in https://github.com/hypriot/flash/issues/177#issuecomment-636901667. I got the same results for all the commands.

firecyberice, 2020-07-21:
@rafaeleyng Thanks for your detailed report. Things evolved over time and the FAQ and docs got adapted from different persons as well. With that said sorry for the inconsistency.

For clarification of your **hostname - fourth way** section

> This works white_check_mark. And by that, I mean: this did set the hostname (I'm not mentioning wifi so far).
> 
> **I don't understand why**. It seems that I've only added user information and wifi configuration, and that somehow makes the hostname work. I just give up investigating the `hostname` at this point.

When you add a `cloud-init.yaml` file without a user block like:
```
# You could modify this for your own user information
users:
  - name: pirate
    gecos: "Hypriot Pirate"
    sudo: ALL=(ALL) NOPASSWD:ALL
    shell: /bin/bash
    groups: users,docker,video
    plain_text_passwd: hypriot
    lock_passwd: false
    ssh_pwauth: true
    chpasswd: { expire: false }
```
... then no user gets created so you cannot login.

That's why we added a default cloud-init userdata file if no specific userdata file is provided.

I have also struggled with the cloud-init docs in the beginning but if they are insufficient we should maybe open an issue or even better a PR there?

rafaeleyng, 2020-07-22:
Hey @firecyberice. I've ended up giving up on Hypriot some time after this issue was created. Since I wasn't able to make things work, I'm not comfortable in making a PR (I don't understand the problem and the solution, so I can't fix it).

YasserAntonio, 2020-08-09:
Hello guys just an easy workaround:  When flashing dont set a user-data with flash. Just use flash to copy the Hypriot image on the sd card. After flashing remount the sd card and simply copy past your user-data on it (must be named this way, without extension) and it will work at boot.

bkahlert, 2021-01-12:
@rafaeleyng I feel you
Was just about to run into the same problems... half a year later. Docs as inconsistent as ever.
What you describe matches so much to what I experiences that I start to believe that that's the destiny of the silent majority of first time users.
Well and opening PRs... You would have to have at least some moments of success to feel motivated enough.

TheVegaByte, 2021-02-20:
I did the following steps to change the hostname after run Hypriot 1.12.3 for the first time and it worked for me!

1. Change the hostname value in /boot/user-data.
2. Type the following commands:

    - sudo cloud-init clean
    - sudo cloud-init clean --logs
    - sudo cloud-init clean --reboot

After this the hostname should have changed.

mrcabbie, 2021-06-15:
I made wifi work! read this https://kerneldriver.wordpress.com/2012/10/21/configuring-wpa2-using-wpa_supplicant-on-the-raspberry-pi/

cengbrecht, 2021-08-09:
> I made wifi work! read this https://kerneldriver.wordpress.com/2012/10/21/configuring-wpa2-using-wpa_supplicant-on-the-raspberry-pi/

From what I can read of this github issue, and the resultant conversation, nothing has changed, and the wifi part is only 1 very small part of the whole issue. 
I too have NOT been able to change the username or the whole thing breaks again, and I have reflashed, and tried many different methods to get the default user to be something else. 
ANY CHANGES that I make to the initial user-config break all ability to login, even though I have not removed or added ANY spaces whatsoever.

cengbrecht, 2021-08-09:
> I did the following steps to change the hostname after run Hypriot 1.12.3 for the first time and it worked for me!
>
> 1. Change the hostname value in /boot/user-data.
> 2. Type the following [...]

I have three pi's running Hypriotos, and am testing this RIGHT now. 
Boot process is taking a bit longer. 
Well, it did not undo my naming schema, however, I will have to re-reflash them again to know. For now, I will do it on the master only. 
Can confirm that this whole issue, and the documentation still suck.

closed 2023-05-14 by rafaeleyng