hypriot/flash #196: Nothing happens

cmarquezrusso, 2021-03-20:
I am trying to install ubuntu on a sdcard for a rpi. 

When I run flash, the copy process doesnt work. The SDCARD is empty
```
$ time flash --userdata cloud-config.yml ubuntu-20.04.2-preinstalled-server-arm64+raspi.img.xz
/usr/local/bin/xz
Uncompressing ubuntu-20.04.2-preinstalled-server-arm64+raspi.img.xz ...
Use /tmp/image.img

Is /dev/disk2 correct? yes

real	1m14.850s
user	0m52.660s
sys	0m7.555s
$
```

System info:
Darwin Cristians-MBP 20.3.0 Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu 7195.81.3~1/RELEASE_X86_64 x86_64

MacOs Big Sur - 11.2.2 (20D80)

tinsjourney, 2021-03-29:
Output of command "diskutil info" has changed in the last version of Big Sur.
On line 203 of flash script, it's looking for "Read only-Media", but now the output is "Media Read-only".

as quick workaround you can still change this in the script, while it's fixed upstream.

tinsjourney, 2021-03-29:
correction it's fixed in master branch but it's not released yet

xUnholy, 2021-05-23:
Any idea what version this fix will be release in?

StefanScherer, 2021-06-02:
I've published version 2.7.2. Hope that helps 👋

closed 2021-06-02 by StefanScherer