hypriot/flash #206: Is this repo still being maintained?

davedittrich, 2022-09-19:
Before I submit yet another PR for changes, I wanted to reach out and see if this repo is still being actively maintained (or as been abandoned). The latest change was to remove mention of donations for support and several issues and PRs go back to 2019.

Some of the PRs are for things I have already added to my own fork and would be better off in the `master` branch. I also have fixes for several Issues and I just added the ability to insert multiple files by repeating the `--file` argument. Ideally, there is a lot of improvement that could be made to `flash`, but it will need some active engagement with a pirate or two who own the repo. :)

iptizer, 2022-09-25:
@StefanScherer @firecyberice Would be great if you could try to find some new maintainers for these projects, would be sad with all that has been achieved so far. Let me know, I would even volunteer to review PRs and take care of such stuff.

davedittrich, 2022-09-27:
I, too, could put in a bit of time to review PRs and see which Issues I could turn into PRs, if that would help.

I know how much time goes into maintaining FOSS projects (and I don't want to over-commit, as I'm supporting myself at the moment developing my own code). This is a general problem, unfortunately.

[Who Will Pay the Piper for Open Source Software Maintenance?](https://www.usenix.org/system/files/login/articles/login_summer20_11_geer.pdf), DAN GEER AND GEORGE P. SIENIAWSKI, Usenix `;login:`, SUMMER 2020

That said, I am using other FOSS components in my projects (`flash` being one of them) and would rather give back to a community-driven project instead of maintaining a fork just to get or fix features I need.